FreeCAD/FreeCAD #19151: [Draft/BIM] Undo/Redo hotkeys for transformation tools do not work (with "continue" setting active)

kaiwas, 2025-01-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In the Draft and BIM workbenches, the key combinations for the transformation tools do not work. However, Undo/Redo work via the interface buttons.
Demonstration on video.

https://github.com/user-attachments/assets/ac8bd1ae-fe22-4e5b-86a3-d4098be4cbbd

https://github.com/user-attachments/assets/69e4eeed-4cc1-44ab-a76c-1f18eb898d6b

Perhaps the problem manifests itself elsewhere.

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39896 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * FreeCAD_SketchArch
  * lattice2 1.0.0
  * PitchedRoof
  * Road 2025.1.16
  * addFC 1.1.0
  * freecad.gears 1.3.0
  * sheetmetal 0.7.3
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-01-20:
I am able to reproduce the described behavior.

When you close the command before invoking the Ctrl+Z, the `Undo` command works correctly. 

Ctrl+Z shortcut does not register, when a command is still active.

maxwxyz, 2025-01-20:
@Roy-043 FYI, is this a bug?

Roy-043, 2025-01-20:
I think this is a duplicate of #6699.

semhustej, 2025-01-21:
> I think this is a duplicate of [#6699](https://github.com/FreeCAD/FreeCAD/issues/6699).

Yes, it seems like it is.

closed 2025-01-21 by maxwxyz