FreeCAD/FreeCAD #19152: Undo/redo behavior in the new transformation tool is strange.

kaiwas, 2025-01-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Undo/redo works very strangely in the new transformation tool.

We get different results depending on the number of transformation types.
That is, "Move" and "Move+rotate" give different undo results.

Sometimes the object resets to its original position, sometimes undoing the action results in its deletion (the object creation step is triggered)
Video demonstration.

https://github.com/user-attachments/assets/fd31638f-c94e-4db9-aa3b-908ef06bc9df

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39896 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * FreeCAD_SketchArch
  * lattice2 1.0.0
  * PitchedRoof
  * Road 2025.1.16
  * addFC 1.1.0
  * freecad.gears 1.3.0
  * sheetmetal 0.7.3
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-20:
@kadet1090 could you take a look at this?

kadet1090, 2025-01-20:
This is due to https://github.com/FreeCAD/FreeCAD/issues/19139. 

Generally for now we have two options:
1. We define transaction to be each movement, but the transactions are hard to define for keyboard inputs. Another problem is that after you confirm you still will have to undo each small movement.
2. Transaction is large and only contains the total movement so you can undo it as one. This is the current state.

I don't see an issue with changing from 2 to 1 but this is only changing one set of problems to another set of problems.

kaiwas, 2025-01-20:
> 2\. Transaction is large and only contains the total movement so you can undo it as one. This is the current state.

It is better to try to save ANY user action. But I understand that this can be difficult.

A long time ago, the behavior of point 2. was in blender. Actions in object mode did not take into account changes in edit mode (v2.79). And it often happened that the user lost data in one undo. Even if it was 10 minutes of work in object mode.

Fortunately, this problem is now fixed (v. 4.3)

Video demonstration.

https://github.com/user-attachments/assets/304037fe-e87a-4939-9e58-62138f1e9b1a

Any loss of user actions is always bad. At the same time, the user should have a way to easily and quickly rewind operations back and forth. This is very useful when designing something new (and not just when building on already attached references).

kadet1090, 2025-01-20:
I agree and I understand the proper behaviour. Ideally as long as you are in the tool you should be able to undo each small movement, only when you confirm it should be squashed and treated as one. This however is not really possible right now. I'll look into it but this seems like a big issue in FreeCAD in general

belinski-roman, 2025-01-20:
> This is due to https://github.com/FreeCAD/FreeCAD/issues/19139.
>
> Generally for now we have two options:
> 1. We define transaction to be each movement, but the transactions are hard to define for [...]

The local movement counter is reset when the coordinate system is rotated or changed. This can be taken as a starting point for cancellation actions. It is only important that you can use Ctrl+Z during the operation of the transformation tool.

kadet1090, 2025-01-20:
> It is only important that you can use Ctrl+Z during the operation of the transformation tool.

Disagree, it is also important that you can undo the whole transformation after you for example click "cancel" or ctrl+z after clicking "OK"

kadet1090, 2025-01-20:
Let's wait for more opinions - if more people want to have behavior with each step being registered for undo (which can easily overflow the undo stack as it has finite and not really big capacity resulting in discarding older) I can make the change. In the meantime I'll think how to achieve the more correct solution described in that comment: https://github.com/FreeCAD/FreeCAD/issues/19152#issuecomment-2602751026

maxwxyz, 2025-01-20:
More important IMO is to undo/redo the complete transform (after you have clicked ok). When the tool is open, you could always cancel, and if you move or input values, they are still in the task dialog (last ones before reset).

maxwxyz, 2025-01-27:
Is this the same as https://github.com/FreeCAD/FreeCAD/issues/18914 ?

kaiwas, 2025-01-27:
> Is this the same as [#18914](https://github.com/FreeCAD/FreeCAD/issues/18914) ?

I can't agree that this is a duplicate.
That topic describes the visual position of the manipulator. This issue specifies undo/redo operations and does not address the issue of the manipulator position at all.

belinski-roman, 2025-01-27:
> Давайте подождём других мнений — если больше людей захотят, чтобы каждый шаг регистрировался для отмены (что может легко привести к переполнению стека отмены, так как он имеет конечную и не очень большую ёмкость, что приводит к удалению более старых) я могу внести изменения. Тем временем я подумаю, как добиться более правильного решения, описанного в этом комментарии: [#19152 (комментарий)](https://github.com/FreeCAD/FreeCAD/issues/19152#issuecomment-2602751026)

if we store micro cancellations during the operation of the tool and overwrite them irrevocably with the total movement after pressing OK, then we do not overflow anything. But I could be wrong.

kadet1090, 2025-01-27:
Yes, that is what I am suggesting, but it needs to be implemented as for now FreeCAD does not have such mechanisms