FreeCAD/FreeCAD #19153: [Draft/BIM] The "continue" setting affects the creation of objects (walls/columns, etc.).

kaiwas, 2025-01-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

There is no "continue" setting for the "wall/beam/column" tools (and possibly others). However, this setting is activated in the "move/rotate/scale" tools.
Video demonstration (using columns as an example).

https://github.com/user-attachments/assets/bc46cde7-97e7-4c6d-90f5-4bc7e13a76af

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39896 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * FreeCAD_SketchArch
  * lattice2 1.0.0
  * PitchedRoof
  * Road 2025.1.16
  * addFC 1.1.0
  * freecad.gears 1.3.0
  * sheetmetal 0.7.3
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-01-20:
It seems to me the `Continue` option is available in `Wall options` or `Structure options` part of the task panel:

[image]

kaiwas, 2025-01-20:
> It seems to me the `Continue` option is available in `Wall options` or `Structure options` part of the task panel:

Hmm. Indeed. I was looking for this checkbox in the same place where it is located in the transformation tools.
Probably this is a design error.

Taking this information into account, the issue can probably be closed. But the interface design looks strange.

kaiwas, 2025-01-20:
> So it is just missing for beams and columns? Could you edit the title then?

As it turns out, these checkboxes exist. But it all works in a very strange way.
The checkbox is not located where I expected to see it.
And it seems that "Continue" works for the entire workbench as a whole. That is, you can turn on or off the checkbox in the wall and have the same result in the beam or in the turn.

I don't know what is the best way to proceed. I can correct the entire issue as a whole (it needs a completely different meaning than the original). Or should I delete this one and create a new one?

kaiwas, 2025-01-20:
Here everything clearly looks wrong somehow. either all the checkboxes are on the left. or all on the right. But it's not good when both are on the left. And also in different places. That's why I didn't notice it.

[image]

kaiwas, 2025-01-20:
> So it is just missing for beams and columns? Could you edit the title then?

I have corrected the first message. But now it includes two questions.
1) The checkbox placement is not logical.
2) The checkbox behavior is not logical.

If I should separate these two questions, please let me know.

semhustej, 2025-01-20:
1. I agree that changing the position of `Continue` checkboxes under the Global option would be probably less confusing for users, as it is there with other tools. On the other hand, it should not be in the `First point of the wall` part of the command,` Wall options` seems like a more logical placement of this option. BUT, when you are choosing the next point of the wall, the Continue option actually shows up there :). It is in the task panel twice. So it makes sense it should be there also when choosing the first point and not be in the `Wall options` part. The same happens with `BIM_Beam` tool.

[image]

2. Not setting `Continue` mode globally, but for each tool independently makes sense to me. I would only want to use this option on a few selected tools, but I do not want to switch or check this option each time I start a tool. I must confess it slows me down in my workflow if I use this option and I mostly have it turned off.

kaiwas, 2025-01-20:
> 1. BUT

very curious)
in the column the top checkbox does not appear. because the column is set to the first point.

kaiwas, 2025-01-20:
There are two possible ways:
1 way - fast (temporary). Make a checkbox "Continue" at the top, visible under any conditions. Make a mark (globally), so that users know that it affects all tools.
2 way - long (correct). Make a checkbox "continue" at the top, visible under any conditions. Without marks, but working separately for each tool.

semhustej, 2025-01-20:
> (remove the checkboxes at the bottom of the panel, because the top part is open and visible always, and the bottom can be hidden, for example, at a low screen resolution).

Yes, this makes sense to me.

closed 2025-04-18 by Roy-043