FreeCAD/FreeCAD #19155: Draft_Dimension Continue mode

semhustej, 2025-01-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Feature request:
Most of FreeCAD Draft/BIM tools have the `Continue` option, which is set globally for all tools. When `Continue` is turned on the command starts again automatically after it is finished the first time.

`Draft_Dimension` has the `Continue` option too, but it alters the functionality of the basic command. Instead of starting the command again after it is finished, the `Continue` option switches the `Draft_Dimension` command into creating continuous dimension. This is not consistent with other tools and may cause confusion for the user.

**Proposed solution:**

- `Continue` option should work as with any other tool - end the creation of the first dimension and start creation of second dimension independent of the first one. This way, the option will be consistent across FreeCAD tools.

- New option `Continuous dimension mode` should be made available. This mode will switch the Draft_Dimension tool into the continuous mode. This is currently achieved by the `Continue` option.

`BIM_DimensionAligned`, `BIM_DimensionHorizontal` and `BIM_DimensionHorizontal` should be changed the same way.

Proposed task dialogue change:

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39896 (Git) Conda
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

Draft

### Anything else?

The reason for the change:

- Consistency across FreeCAD tools
- When working with the Draft/BIM WB, I often use continuous dimensions. When switching `Continue` option on, I have to repeatedly turn it of when using other tools (where I don't use the Continue option) and than turn it back on when creating continuous dimension. I think I might not be the only one who needs to access the continuous dimension often when creating AEC documentation
- Better exposure of the continuous dimension option

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

kaiwas, 2025-01-20:
> Most of FreeCAD Draft/BIM tools have the `Continue` option, which is set globally for all tools.

This is actually very bad. This option should work separately for each tool and save its value separately.
Or at least it should have an explicit mark about its globality.

For example, I may want to create 10 columns, but I rotate them more often once. The frequency of using "continue" for each tool is different and depends on the frequency of use.

I completely agree about "continue" related to sizes.

semhustej, 2025-02-18:
> 
> About the different continue mode for dimensions: Indeed this is confusing. [@semhustej](https://github.com/semhustej) 's proposal would indeed be a solution. Maybe the current "continue" mode should be renamed to "restart command"?

The naming is tricky.
"Restart command" sounds like a standard "Continue" mode to me. 

I would choose either "Continuous mode" or "Continued dimension". I am heavily influenced by years of using AutoCAD, where these terms are used.

[image]

https://help.autodesk.com/view/ACD/2023/ENU/?guid=GUID-27CF5248-EECB-4F5E-9A25-39ED399EA1EC

semhustej, 2025-02-19:
> Chained dimension sounds good, it's used elsewhere too. Only problem, we're just creating normal dimensions side-by-side, not actual chained dimensions, that can be edited in one go. Maybe that's something we should implement too, but it's more work and I'm not fully sure of all the characteristics it should have...

That's truth. So should it be named "Chained mode"? It is not a chained dimension, but only a 'chained mode' of the standard dimension. 
If a true chained dimension (as one object or dependent objects) would be developed, it could be named 'Chained dimension'.

Or we could name it "side by side" but that does not sound too professional to me :)

yorikvanhavre, 2025-02-20:
"Chained mode" seems perfect!

tetektoza, 2025-04-06:
I will try to take this one on in the following days to get more familiar with the codebase, seems like a good beginner issue. So basically, there are two points to be done and let me sum them up so I will know on what you guys actually have decided:
1. make "Continue" option independent for each tool, so if one checks the checkbox, another one won't have it selected by default
2. add additional option for Dimensions, which would basically transfer current "Continue" workflow to it.

Am I right? @yorikvanhavre @semhustej

semhustej, 2025-04-07:
> 
>     1. make "Continue" option independent for each tool, so if one checks the checkbox, another one won't have it selected by default

Yes. The `Continue` state should be independent for each Draft tool. There is quite a lot of tools, which have this option. Let me know if it would be helpful to you if I went through FreeCAD and listed the tools here. I am not sure if there is a 'Class' for the continue option or if it is necessary to edit each one.

> 
>     2. add additional option for Dimensions, which would basically transfer current "Continue" workflow to it.
> 

Yes. Basically the current `Continue` can be just renamed to `Chained mode` and a new option called `Continue` created. This new `Continue` will basically just start the command all over again.

Thanks for your input. I am not a developer, Yorik might give you some pointers.

tetektoza, 2025-04-07:
@kaiwas Thanks. I think you're talking about https://github.com/FreeCAD/FreeCAD/issues/19153 am I right? I think both of those issues have interlapping ideas, so I will also try to handle the part about moving the checkbox above. I think the basic idea from the other issue reported was:
> way - long (correct). Make a checkbox "continue" at the top, visible under any conditions. Without marks, but working separately for each tool.

Working separately for each tool will be done in this issue, but if I'll manage moving the checkbox, we can close the second issue as well.

kaiwas, 2025-04-07:
> Thanks. I think you're talking about [#19153](https://github.com/FreeCAD/FreeCAD/issues/19153) am I right?

Yes, you are right.
Thanks in advance! It would be great if this setting would look the same everywhere. Uniformity is a definite benefit.
I wish you success!

tetektoza, 2025-04-07:
BTW, probably question to you @kaiwas - before I change it - I currently see, that `continueMode` is being saved to user.cfg through `param_dict["Mod/Draft"]`, so once user selects it, it's selected in the next run of the app. Should I add different continueMode (or chainedMode as it's soon to be changed) for all the tools that we support so they will have their own value cached and for every single tool it will be remembered differently, or do we want to remove this variable at all? Couldn't find an example where variable is cached separately for different tool, as it seems most UI parts are "global", but they are just not shown for different tools, depending on the context.

kaiwas, 2025-04-08:
> BTW, probably question to you [@kaiwas](https://github.com/kaiwas)

For me, this is a completely unnecessary setting that rather gets in the way. In my opinion, all tools should always work in the "continue" mode. Because if I no longer need the tool's action, I will simply close it.
That is, I would make "continue" by default for absolutely all tools without the ability to disable it.

But I did not come up with such logic and it is not for me to decide.

tetektoza, 2025-04-08:
@semhustej No problem, I have like 80-90% implementation done, just need to do testing and I think today or tomorrow I will post a PR. I've figured a pretty clean way on how to store those settings for each tool separately in config.cfg.

semhustej, 2025-04-08:
> [@semhustej](https://github.com/semhustej) No problem, I have like 80-90% implementation done, just need to do testing and I think today or tomorrow I will post a PR. I've figured a pretty clean way on how to store those settings for each tool separately in config.cfg.

Great, that was fast! If Roy does not respond in this thread I guess we will see in the PR discussion :).

Roy-043, 2025-04-08:
> store those settings for each tool separately in config.cfg.

I have not followed this discussion, and do not know exactly what your plans are, but you should use the existing parameter system to store these setting.

tetektoza, 2025-04-08:
@Roy-043 My thought was to add another param group under "Draft" (and later on BIM), so instead of looking like this:

```
        <FCParamGroup Name="Draft">
            <FCInt Name="gridTransparency" Value="0"/>
            <FCUInt Name="constructioncolor" Value="746455039"/>
            <FCUInt Name="gridColor" Value="1230002175"/>
            <FCUInt Name="snapcolor" Value="4289331455"/>
            <FCText Name="textfont">Arial</FCText>
            <FCBool Name="ContinueMode" Value="1"/>
          </FCParamGroup>
```

It will look like this (ContinueMode (or ChainedMode as it's soon to be changed), stored separately for each tool):

```
        <FCParamGroup Name="Draft">
            <FCInt Name="gridTransparency" Value="0"/>
            <FCUInt Name="constructioncolor" Value="746455039"/>
            <FCUInt Name="gridColor" Value="1230002175"/>
            <FCUInt Name="snapcolor" Value="4289331455"/>
            <FCText Name="textfont">Arial</FCText>
            <FCParamGroup Name="ContinueMode">
              <FCBool Name="Dimension" Value="1"/>
              <FCBool Name="Arc" Value="1"/>
              <FCBool Name="Line" Value="1"/>
            </FCParamGroup>
          </FCParamGroup>
```

What do you think?

Roy-043, 2025-04-09:
Your reference to config.cfg confused me. AFAIK the FreeCAD file is named user.cfg on all platforms.

tetektoza, 2025-04-09:
@Roy-043 🤦‍♂ Yep, you're right. Sorry, brain fog happened here :D

Roy-043, 2025-04-09:
> Couldn't find an example where variable is cached separately for different tool,

Currently all Draft and BIM parameters are in one group per workbench. To introduce a separate group for this purpose makes sense. The code in params.py does not account for this situation (I think), but you should not let that stop you from creating a subgroup.

closed 2025-04-18 by Roy-043