FreeCAD/FreeCAD #19156: [Draft/BIM] The information icon in the fields has an extra state and is not aligned correctly.

kaiwas, 2025-01-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The gif image shows that the information icon has three states.
1. Correct data (green checkbox)
2. Incorrect data (red prohibition sign).
3. No icon (if the field is empty).

[image]

In addition, the alignment of icons along the right edge in some tools is not the same. For three signs, we have three different indents from the right edge.

[image]

1. Correct.
2 + 3. Incorrect. Because an empty field can be considered an incorrect input.

The appearance of the icons can also be made more modern.

For example, this is how it looks in Bootstrap v.5.3.

[image]

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39896 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * FreeCAD_SketchArch
  * lattice2 1.0.0
  * PitchedRoof
  * Road 2025.1.16
  * addFC 1.1.0
  * freecad.gears 1.3.0
  * sheetmetal 0.7.3
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

yorikvanhavre, 2025-01-20:
Note that this is not only Draft but anything that uses quantity widgets.

maxwxyz, 2025-01-21:
You are using a third party theme. How does it look with the themes classic, dark, light? It's the scale and orientation better?

kaiwas, 2025-01-21:
> You are using a third party theme. How does it look with the themes classic, dark, light? It's the scale and orientation better?

The indents vary depending on the tool. For some it's fine. But most often the first icon is pressed to the right.

but alignment is the least of the problems. I am currently testing themes and it turned out that the "incorrectness" icon is not always shown. At the same time, the dimension sometimes disappears.

Freecad classic

[image]

FreeCAD Dark 

[image]

FreeCAD Light

[image]

Open light

[image]

If I enter Cyrillic characters, sometimes the icon shows the correct data.
Sometimes millimeters disappear. Sometimes they reappear. Everything looks somehow "random".

kaiwas, 2025-01-21:
I think the most correct solution was to always display the dimension (for example, mm). In order to select and correct only numbers, and millimeters would simply always be shown and would not be selectable.

[image]

Well, and design the fields differently. The current icons behave incorrectly and look like dinosaurs.

maxwxyz, 2025-01-21:
@MisterMakerNL FYI

PR #20466 (2025-03-30): Gui: fix Inputfield valid/invalid icon placement and size
fixes #19156

this is the difference between master and this PR:
[image]

here is an overlap of this PR comparing it to the fx icon in master:
[image]
closed 2025-03-30 by kadet1090