FreeCAD/FreeCAD #19157: Assembly: Large assembly fails

jkvfc, 2025-01-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

[image]
If the brown spring washer is added to the large assembly above, the following error message appears:
```
13:56:59  2e-08 <Assembly> AssemblyObject.cpp(176): Solve failed: vector::_M_range_check: __n (which is 500) >= this->size() (which is 500)
```
The first spring washer and the components before it could be added without any problems.
Once added, the Refresh command works any longer and displays:
```
14:00:56  236.703 <App> Document.cpp(2903): SpritzwzR#Joint still touched after recompute
14:00:56  236.703 <App> Document.cpp(2903): SpritzwzR#Joint001 still touched after recompute
14:00:56  236.703 <App> Document.cpp(2903): SpritzwzR#Joint002 still touched after recompute
...
```
The Solve Assembly command returns the first error message mentioned above.
The assembly file is:

[Spritzwz.zip](https://github.com/user-attachments/files/18478886/Spritzwz.zip)

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39868 (Git)
Build type: Unknown
Branch: main
Hash: a66da9b3d2281c5ef0a20ce14bacaf21f66f31f3
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ftdt10, 2025-01-29:
My solutions for "still touched after recompute" errors and performance issues
1. Wrapping with [Assembly container](https://wiki.freecad.org/Assembly_CreateAssembly), especially objects imported from step files.
2. When adding multiple identical parts to an assembly, use the [Link Array](https://wiki.freecad.org/Std_LinkMake#Link_Array) function instead of Copy-Paste or Link.

jkvfc, 2025-01-29:
@ftdt10 Thank you very much for your advice but unfortunately sub-assemblies and links are poorly supported by the new assembly workbench introduced in version 1.0.0. I was able to add the four spring washers but if the next item (screw) is added the above mentioned 'Solve failed' error is still there. I assume you meant I am using the old assembly workbench. In my opinion there is a deeper problem in the new assembly workbench.

ftdt10, 2025-01-30:
Still touched error for a Link object is also solved by duplicating it instead of using Link in my project testing.