FreeCAD/FreeCAD #19160: TechDraw: Page in focus blocks theme change and first start setup

maxwxyz, 2025-01-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When I have a TechDraw page open, I cannot change the FreeCAD theme in the preferences. The dropdown is greyed out.
This only happens when I start the preferences from a TechDraw page view. Any other 3D view works as expected.

[image]

I also noticed this: When I open the Start page (Help>Start) from an active TechDraw page view, the open first start setup button is greyed out and I cannot click it:

[image]

This button is not greyed out, when starting it from any other 3D view.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39896 (Git) Conda
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.29.4
  * CurvedShapes 1.0.13
  * Curves 0.6.53
  * dodo 1.0.1
  * fasteners 0.5.35
  * freecad.gears 1.3.0
  * FreeCAD_SketchArch (Disabled)
  * kicadStepUpMod 11.5.2
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * parts_library
  * parts_library.backup1737145514.009932
  * Rocket 4.0.1
  * sheetmetal 0.6.11
  * Silk 0.1.5
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-01-20:
@maxwxyz a hot fix was implemented to stop a crash which no-one had been able to get to the root cause of: https://github.com/FreeCAD/FreeCAD/pull/17152

maxwxyz, 2025-01-20:
sounds not very good :)