FreeCAD/FreeCAD #19171: No way to reset individual preferences / display which preference was changed by the user or theme

maxwxyz, 2025-01-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Currently, the user can only reset all preferences, the preferences from the tab or the group. There is no way to reset individual preferences.
There is also no way to see, which preferences were changed in comparison to the default value.

I propose a similar I as in existing apps, where any changed preference is displayed in bold with a reset button next to it. Hovering the reset button would show the default value, clicking it would only reset this preference field.

Similar UI in Firefox:

[image]

It would be great to have some sort of information on changes based on a different setting: For example, many TechDraw preferences changes when you switch from ASME to ISO. This relation, why a preference changed and because of which setting it has been changed (and the default when choosing the parent setting) would be great.

 

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39896 (Git) Conda
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * Curves 0.6.51
  * fasteners 0.5.32
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

hyarion, 2025-01-21:
Isn't this a duplicate? I'm quite sure we've discussed this in an earlier GH issue (could have been on discord)

The problem as I see it is that FreeCAD doesn't know if a setting comes from a preference pack, is default or user changed.

What I think we need to be able to solve this issue to implement preferences as a stack:
1. first apply defaults
2. preference packs (multiple, in the order installation)
3. user settings

When that's done it would be possible to reorder preference packs, clear user settings and show what different stack items values are.

maxwxyz, 2025-01-21:
@hyarion I also thought so but did not find an issue.
At least we should know the default value (as there is a possibiity to reset the preference page) and check if the current value is different from the default?

hyarion, 2025-01-21:
> At least we should know the default value (as there is a possibiity to reset the preference page) and check if the current value is different from the default?

We can clear the value, but not inspect it. The problem is that the default values are specified where the data is read. In fact, we can have multiple default values depending on where we read a cleared/unset value.

kaiwas, 2025-01-21:
I had an issue about resetting values. But your version looks more global.

https://github.com/FreeCAD/FreeCAD/issues/17685

furgo16, 2025-01-21:
If we were to have a UI similar to Firefox (as seen in the issue description screenshot), it might be worth showing both preferences and [parameters](https://wiki.freecad.org/Fine-tuning) there.

maxwxyz, 2026-02-07:
From the 2026 complaint session:
**We need a way to get a listing of options that are not set to their defaults**