FreeCAD/FreeCAD #19172: Draft: Facebinder incorrectly updated

rostskadat, 2025-01-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

## Context

- I have an `Arch::Wall` that is created with a `Facebinder` on some of the faces.
- The `Facebinder` is created using `Draft.make_facebinder(( wall, ("Face2", ) ))`
- At some point I add a first and a second `Arch::Window`.

Suddenly the `Facebinder` fails to properly update and take the second window into account.

## Diagnostic

It seems that the attribute `Facebinder.Faces` (in `FreeCAD/src/Mod/Draft/draftobjects/facebinder.py`) is updated to contain
```python
(wall, ("?Face2", ))
 ```
instead of the original
```python
(wall, ("Face2", ))
 ```

Modifying the `Facebinder::execute()` to remove that extra `?` character seems to solve the problem.

## Caveat

- I'm not sure what changes the definition of `Facebinder.Faces`  nor why the extra `?` character is added (¿could it be related to the TNP?)

## Sample

The following files demonstrate the problem

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (xcb)
Architecture: x86_64
Version: 1.1.0dev.39898 (Git)
Build type: Debug
Branch: main
Hash: 1b8bac38057469af3221bebb4974cc2b01113cd2
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-27:
@rostskadat is this fixed by https://github.com/FreeCAD/FreeCAD/pull/19178 because this was mentioned in those commits?

rostskadat, 2025-01-27:
Hello @maxwxyz ,

Nope, it is not fixed, I have changed my code to implement a mitigation.
The mitigation consists in trying to determine the proper face after all doors&windows have been added to the wall.
This is not a workaround the described behavior. 
Ideally the face name set by the user would not change even after adding a door & window to the wall.

I understand that might be difficult :)

yorikvanhavre, 2025-02-06:
Not sure why the getSelectionEx() now returns things like `?Face2`, it seems to have happened after the TNP work. In any case your fix seems good. Mind to make a PR? 😇

rostskadat, 2025-02-07:
@yorikvanhavre I can do a PR, however don't you think that would just hide the underlying problem? If the change that happens in the `Facebinder.Faces` attribute should not happen (no idea on that unfortunately), maybe it is a symptom more than the root cause. What do you think?

FEA-eng, 2025-11-03:
Recomputing this document gives me errors:

```
11:21:49  wall0-fb-left: Unable to build facebinder
11:21:49  wall0-fb-right: Unable to build facebinder
```

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.20251102 (Git shallow)
Build date: 2025/11/02 16:03:37
Build type: Release
Branch: main
Hash: 45ca23d074f39a9b586c8320f1e7cd20ca466fef
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: unset/FreeCAD Light/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```