FreeCAD/FreeCAD #19173: Draft: Snap Tracking

semhustej, 2025-01-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

**Feature request:**
When trying to place or create object (by selecting a point), which should be a certain (let's say 1000mm) orthogonal distance form existing snap point, it is currently not possible in FreeCAD Draft. User has to resort to two step process - 
1. place or create the object at the snap point
2. Draft_Move the object 1000mm in the orthogonal direction

Standard DWG editors have tool which enables orthogonal tracking from a snap point to a user-specified distance. See examples from BricsCAD

1. Task - create a new line with starting point 1000mm right of the rectangle corner:

https://github.com/user-attachments/assets/d616cbf1-3d40-4b44-8565-1772d1d85863

2. Task - move the door object 500mm to the bottom of the corner

https://github.com/user-attachments/assets/eab1a9b6-2a3d-4b07-a7cb-009e53a1902a

As seen in the examples, with snap tracking these tasks are a one-step (command) process. With current FreeCAD Draft Tools, this would be two-step (commands) process.

**Proposed feature:**
The proposed feature could use the 'Hold point' (Q) as a basis (as some of the snap tracking functionality is already there), or it could be independent of this regime (different keyboard shortcut). If user would create such a 'Hold point', the distance and position of the new point would not be measured form the [0,0,0] point of the coordinate system, but form the hold point. Preferably in X, Y and Z directions.
When several 'Hold points' are created, the distance should be measured from the last one or the first one.

Example process for task  - create a new line with starting point 1000mm right of the rectangle corner:

- Start a line tool
- Create 'Hold point' at the corner of the rectangle
- Move mouse cursor to the right so that the tracking direction is highlighted
- Type '500' and hit enter on keyboard

This would place the first point of the new line 500 units right of the rectangle corner.

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39896 (Git) Conda
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

Draft

### Anything else?

In my opinion, this feature would greatly enhance modelling speed, both in 2D and in 3D. Having desired placement at a certain distance from a corner or another important snap point is used very often in AEC modelling.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-22:
@Roy-043 is this a complex implementation?

yorikvanhavre, 2025-01-22:
I have two ideas here:

1.  When hovering the mouse on a hold point, the "last point" tracked by the UI switches to that hold point automatically, OR:
2. We introduce a new shortcut key that says: "consider the current snap point under the cursor to be the "last point" tracked by the UI". This is less automatic, but would allow more possibilities, such as not needing a hold point at all (one could just use any snap position)

I think both are relatively easy to achieve, one should check how the snapper could override the "last point", keep it across snapper launches, and inform a correct final position when the operation ends. I think it can all be done inside the snapper, we basically need a "last point override" mechanism.

semhustej, 2025-01-22:
Thanks for considering the feature.

> 1. When hovering the mouse on a hold point, the "last point" tracked by the UI switches to that hold point automatically, OR:

This is how it is implemented in BricsCAD/AutoCAD. 

> 2. We introduce a new shortcut key that says: "consider the current snap point under the cursor to be the "last point" tracked by the UI". This is less automatic, but would allow more possibilities, such as not needing a hold point at all (one could just use any snap position)

I think this might be the more robust way of implementation. Even though not as fast solution for experienced user as option 1, I think this would avoid potential pitfalls of option 1 (such as the 2nd video I recorded, were mouse moves over the line). Worth considering whether it would not be possible to add to the Q-hold point functionality.

kaiwas, 2025-01-23:
> 1. When hovering the mouse on a hold point, the "last point" tracked by the UI switches to that hold point automatically,

There are already enough options with automatic tracking. It only causes trouble.
My opinion is that screen coordinates capture should be controlled everywhere. Simply because the automatic option is unlikely to be implemented in a convenient way.
And the current system needs to be completely redesigned. The improvements made to the Draft workbench should be distributed virtually everywhere.
Loss of entered data from mouse movement is one of the most inconvenient features of FreeCAD.

yorikvanhavre, 2025-01-23:
Indeed the "manual" way (option 2) is definitely the best path... Also because it's also easier to add an automatic system on top of it later if needs be, than the contrary

GS-GOAT, 2025-02-12:
@yorikvanhavre Please review.

Roy-043, 2025-10-23:
Yorik has added the working plane Recenter option (#19728). IMO that covers what is requested here. I therefore suggest that this issue can be closed.

closed 2025-10-23 by maxwxyz

Roy-043, 2025-11-21:
I have tried to explain it here:
https://wiki.freecad.org/Draft_Snap#Advanced_snapping

semhustej, 2025-11-22:
> I have tried to explain it here: https://wiki.freecad.org/Draft_Snap#Advanced_snapping

Thank you!

I did few basic tests and the combination of D + Q + Shift does the job. It is not as slick as AutoCAD solution and needs a bit more steps and attention, but it gets the job done.