FreeCAD/FreeCAD #19174: Crash when opening a file containing a sketch with redundant constraints

jffmichi, 2025-01-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Since a few days FreeCAD crashes in `GCS::System::diagnose(GCS::Algorithm)` when loading the sketch in the following file. 
The file was created with FreeCAD version 1.1.0.39802.g558eb19103 which can also load it without problems.

[sketcher_crash.zip](https://github.com/user-attachments/files/18491530/sketcher_crash.zip)

### Full version info

```shell
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.1.0dev.39935 (Git)
Build type: Release
Branch: makepkg
Hash: 49b304ae558553f9acd5872ccd30673bbd60bd19
Python 3.13.1, Qt 6.8.1, Coin 4.0.3, Vtk 9.4.1, OCC 7.8.2.dev
Locale: English/United States (en_US) [ OS: German/Germany (de_DE) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/breeze
Installed mods: 
  * kicadStepUpMod 11.4.1
  * dodo 1.0.1
  * sheetmetal 0.5.3
  * fcgear 1.0.0
  * fasteners 0.5.29
  * Assembly4 0.50.15
  * Curves 0.6.51
  * lattice2 1.0.0
```

### Subproject(s) affected?

None

### Anything else?

Here is the callstack of the crash:

`0x00007ffff2ea53f4 in ?? () from /usr/lib/libc.so.6
(gdb) bt
#0  0x00007ffff2ea53f4 in ??? () at /usr/lib/libc.so.6
#1  0x00007ffff2e4c120 in raise () at /usr/lib/libc.so.6
#2  0x00007ffff2e334c3 in abort () at /usr/lib/libc.so.6
#3  0x00007ffff30d3af0 in std::__glibcxx_assert_fail (file=<optimized out>, line=<optimized out>, function=<optimized out>, condition=<optimized out>) at /usr/src/debug/gcc/gcc/libstdc++-v3/src/c++11/assert_fail.cc:41
#4  0x00007fff551586c7 in GCS::System::diagnose(GCS::Algorithm) () at /usr/lib/freecad/lib/Sketcher.so
#5  0x00007fff5514969b in GCS::System::initSolution(GCS::Algorithm) () at /usr/lib/freecad/lib/Sketcher.so
#6  0x00007fff551db2aa in Sketcher::Sketch::setUpSketch(std::vector<Part::Geometry*, std::allocator<Part::Geometry*> > const&, std::vector<Sketcher::Constraint*, std::allocator<Sketcher::Constraint*> > const&, int) ()
    at /usr/lib/freecad/lib/Sketcher.so
#7  0x00007fff550bcb6b in Sketcher::SketchObject::setUpSketch() () at /usr/lib/freecad/lib/Sketcher.so
#8  0x00007fff55100511 in Sketcher::SketchObject::onChanged(App::Property const*) () at /usr/lib/freecad/lib/Sketcher.so
#9  0x00007ffff58f5f8c in App::Property::hasSetValue() () at /usr/lib/freecad/lib/libFreeCADApp.so
#10 0x00007fff550c4814 in Sketcher::SketchObject::acceptGeometry() () at /usr/lib/freecad/lib/Sketcher.so
#11 0x00007fff55106dc5 in Sketcher::SketchObject::restoreFinished() () at /usr/lib/freecad/lib/Sketcher.so
#12 0x00007fff550f758e in Sketcher::SketchObject::onDocumentRestored() () at /usr/lib/freecad/lib/Sketcher.so
#13 0x00007ffff57c7621 in App::Document::afterRestore(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool) () at /usr/lib/freecad/lib/libFreeCADApp.so
#14 0x00007ffff57c7ff6 in App::Document::afterRestore(bool) () at /usr/lib/freecad/lib/libFreeCADApp.so
#15 0x00007ffff59917e7 in App::Application::openDocuments(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >*, bool) () at /usr/lib/freecad/lib/libFreeCADApp.so
#16 0x00007ffff59928b1 in App::Application::openDocument(char const*, bool) () at /usr/lib/freecad/lib/libFreeCADApp.so
#17 0x00007ffff59aa601 in App::Application::sOpenDocument(_object*, _object*, _object*) () at /usr/lib/freecad/lib/libFreeCADApp.so
#18 0x00007ffff4d9499c in ??? () at /usr/lib/libpython3.13.so.1.0
#19 0x00007ffff4d5d02c in _PyObject_MakeTpCall () at /usr/lib/libpython3.13.so.1.0
#20 0x00007ffff4d75beb in _PyEval_EvalFrameDefault () at /usr/lib/libpython3.13.so.1.0
#21 0x00007ffff4e40ce5 in PyEval_EvalCode () at /usr/lib/libpython3.13.so.1.0
#22 0x00007ffff4e62870 in ??? () at /usr/lib/libpython3.13.so.1.0
#23 0x00007ffff4e5cc7a in ??? () at /usr/lib/libpython3.13.so.1.0
#24 0x00007ffff4e5068e in ??? () at /usr/lib/libpython3.13.so.1.0
#25 0x00007ffff53179ee in Base::InterpreterSingleton::runString[abi:cxx11](char const*) () at /usr/lib/freecad/lib/libFreeCADBase.so
#26 0x00007ffff61ad507 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) () at /usr/lib/freecad/lib/libFreeCADGui.so
#27 0x00007ffff61ad6a3 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) () at /usr/lib/freecad/lib/libFreeCADGui.so
#28 0x00007ffff60c0260 in Gui::Application::open(char const*, char const*) () at /usr/lib/freecad/lib/libFreeCADGui.so
#29 0x00007ffff6175b85 in Gui::ModuleIO::openFile(QString const&) () at /usr/lib/freecad/lib/libFreeCADGui.so
#30 0x00007ffff61a1f51 in Gui::RecentFilesAction::activateFile(int) () at /usr/lib/freecad/lib/libFreeCADGui.so
#31 0x00007ffff61adbd5 in Gui::Command::_invoke(int, bool) () at /usr/lib/freecad/lib/libFreeCADGui.so
#32 0x00007ffff61ae047 in Gui::Command::invoke(int, Gui::Command::TriggerSource) () at /usr/lib/freecad/lib/libFreeCADGui.so
#33 0x00007ffff35b1b29 in ??? () at /usr/lib/libQt6Core.so.6
#34 0x00007ffff3f644bd in QActionGroup::_q_actionTriggered() () at /usr/lib/libQt6Gui.so.6
#35 0x00007ffff35b1b29 in ??? () at /usr/lib/libQt6Core.so.6
#36 0x00007ffff3f67484 in QAction::activate(QAction::ActionEvent) () at /usr/lib/libQt6Gui.so.6
#37 0x00007ffff46e2bf9 in ??? () at /usr/lib/libQt6Widgets.so.6
#38 0x00007ffff46e4c03 in ??? () at /usr/lib/libQt6Widgets.so.6
#39 0x00007ffff454f816 in QWidget::event(QEvent*) () at /usr/lib/libQt6Widgets.so.6
#40 0x00007ffff44fe78a in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/libQt6Widgets.so.6
#41 0x00007ffff4502a26 in QApplication::notify(QObject*, QEvent*) () at /usr/lib/libQt6Widgets.so.6
#42 0x00007ffff61693bf in Gui::GUIApplication::notify(QObject*, QEvent*) () at /usr/lib/freecad/lib/libFreeCADGui.so
#43 0x00007ffff3557ac8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/libQt6Core.so.6
#44 0x00007ffff44f613d in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) () at /usr/lib/libQt6Widgets.so.6
#45 0x00007ffff4569d62 in ??? () at /usr/lib/libQt6Widgets.so.6
#46 0x00007ffff456ae40 in ??? () at /usr/lib/libQt6Widgets.so.6
#47 0x00007ffff44fe78a in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/libQt6Widgets.so.6
#48 0x00007ffff61693bf in Gui::GUIApplication::notify(QObject*, QEvent*) () at /usr/lib/freecad/lib/libFreeCADGui.so
#49 0x00007ffff3557ac8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/libQt6Core.so.6
#50 0x00007ffff3b8752b in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) () at /usr/lib/libQt6Gui.so.6
#51 0x00007ffff3c02b44 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/libQt6Gui.so.6
#52 0x00007ffff404ea14 in ??? () at /usr/lib/libQt6Gui.so.6
#53 0x00007ffff1bcc559 in ??? () at /usr/lib/libglib-2.0.so.0
#54 0x00007ffff1c2f257 in ??? () at /usr/lib/libglib-2.0.so.0
#55 0x00007ffff1bcba55 in g_main_context_iteration () at /usr/lib/libglib-2.0.so.0
#56 0x00007ffff37c22fd in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/libQt6Core.so.6
#57 0x00007ffff3561ee6 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/libQt6Core.so.6
#58 0x00007ffff3558f4f in QCoreApplication::exec() () at /usr/lib/libQt6Core.so.6
#59 0x00007ffff60ca56c in Gui::Application::runApplication() () at /usr/lib/freecad/lib/libFreeCADGui.so
#60 0x0000555555559166 in ??? ()
#61 0x00007ffff2e34e08 in ??? () at /usr/lib/libc.so.6
#62 0x00007ffff2e34ecc in __libc_start_main () at /usr/lib/libc.so.6
#63 0x000055555555a525 in ??? ()
`

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

xtemp09, 2025-01-21:
> ```ruby
> Build type: Release
> ```

> ```cpp
> #3  0x00007ffff30d3af0 in std::__glibcxx_assert_fail (file=<optimized out>, line=<optimized out>, function=<optimized out>, condition=<optimized out>) at /usr/src/debug/gcc/gcc/libstdc++-v3/src/c++11/assert_fail.cc:41
> ```

How can assert fail in a release build? @jffmichi, where did this version come from?

jffmichi, 2025-01-21:
It's [freecad-git](url=https://aur.archlinux.org/packages/freecad-git) from the Arch User Repository. For me the crash also happens on commit a977fade with vtk version 9.2.6 and opencascade 7.7.2. I'll check what's going on with the debug/release thing.

jffmichi, 2025-01-21:
Here is the callstack with debug information. Crash happes at [GCS.cpp (line 5596)](https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Sketcher/App/planegcs/GCS.cpp#L5596).

```
(gdb) bt
#0  0x00007ffff2ea53f4 in ??? () at /usr/lib/libc.so.6
#1  0x00007ffff2e4c120 in raise () at /usr/lib/libc.so.6
#2  0x00007ffff2e334c3 in abort () at /usr/lib/libc.so.6
#3  0x00007ffff30d3af0 in std::__glibcxx_assert_fail
    (file=file@entry=0x7fff552380b8 "/usr/include/c++/14.2.1/bits/stl_vector.h", line=line@entry=1130, function=function@entry=0x7fff55242e70 "std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = std::vector<GCS::Constraint*>; _Alloc = std::allocator<std::vector<GCS::Constraint*> >; reference = std:"..., condition=condition@entry=0x7fff55235492 "__n < this->size()") at /usr/src/debug/gcc/gcc/libstdc++-v3/src/c++11/assert_fail.cc:41
#4  0x00007fff551574c7 in std::vector<std::vector<GCS::Constraint*, std::allocator<GCS::Constraint*> >, std::allocator<std::vector<GCS::Constraint*, std::allocator<GCS::Constraint*> > > >::operator[] (this=<optimized out>, __n=<optimized out>)
    at /usr/include/c++/14.2.1/bits/stl_vector.h:1130
#5  GCS::System::identifyConflictingRedundantConstraints<Eigen::SparseQR<Eigen::SparseMatrix<double, 0, int>, Eigen::COLAMDOrdering<int> > >
    (this=<optimized out>, alg=<optimized out>, qrJT=..., jacobianconstraintmap=std::map with 15 elements = {...}, tagmultiplicity=std::map with 9 elements = {...}, pdiagnoselist=std::vector of length 14, capacity 16 = {...}, R=..., constrNum=<optimized out>, rank=14, nonredundantconstrNum=<synthetic pointer>: <optimized out>) at /usr/src/debug/freecad-git/FreeCAD/src/Mod/Sketcher/App/planegcs/GCS.cpp:5596
#6  GCS::System::diagnose (this=0x55555df5b718, alg=<optimized out>) at /usr/src/debug/freecad-git/FreeCAD/src/Mod/Sketcher/App/planegcs/GCS.cpp:5181
#7  0x00007fff5514849b in GCS::System::initSolution (this=0x55555df5b718, alg=GCS::DogLeg) at /usr/src/debug/freecad-git/FreeCAD/src/Mod/Sketcher/App/planegcs/GCS.cpp:1829
#8  0x00007fff551d9faa in Sketcher::Sketch::setUpSketch (this=0x55555df5b6d8, GeoList=<optimized out>, ConstraintList=<optimized out>, extGeoCount=<optimized out>) at /usr/src/debug/freecad-git/FreeCAD/src/Mod/Sketcher/App/Sketch.cpp:274
#9  0x00007fff550bbc2b in Sketcher::SketchObject::setUpSketch (this=0x55555df5a1b0) at /usr/src/debug/freecad-git/FreeCAD/src/Mod/Sketcher/App/SketchObject.cpp:1308
#10 0x00007fff550ff591 in Sketcher::SketchObject::onChanged (this=0x55555df5a1b0, prop=0x55555df5af68) at /usr/src/debug/freecad-git/FreeCAD/src/Mod/Sketcher/App/SketchObject.cpp:10211
#11 0x00007ffff58f5f8c in App::Property::hasSetValue (this=0x55555df5af68) at /usr/src/debug/freecad-git/FreeCAD/src/App/Property.cpp:274
#12 0x00007fff550c38d4 in Sketcher::SketchObject::acceptGeometry (this=0x55555df5a1b0) at /usr/src/debug/freecad-git/FreeCAD/src/Mod/Sketcher/App/SketchObject.cpp:1613
#13 0x00007fff55105e45 in Sketcher::SketchObject::restoreFinished (this=0x55555df5a1b0) at /usr/src/debug/freecad-git/FreeCAD/src/Mod/Sketcher/App/SketchObject.cpp:10613
#14 0x00007fff550f660e in Sketcher::SketchObject::onDocumentRestored (this=0x55555df5a1b0) at /usr/src/debug/freecad-git/FreeCAD/src/Mod/Sketcher/App/SketchObject.cpp:10587
#15 0x00007ffff57c7621 in App::Document::afterRestore (this=this@entry=0x5555596206c0, objArray=<optimized out>, checkPartial=<optimized out>) at /usr/src/debug/freecad-git/FreeCAD/src/App/Document.cpp:2413
#16 0x00007ffff57c7ff6 in App::Document::afterRestore (this=0x5555596206c0, checkPartial=<optimized out>) at /usr/src/debug/freecad-git/FreeCAD/src/App/Document.cpp:2352
#17 0x00007ffff59917e7 in App::Application::openDocuments (this=this@entry=0x55555570c9c0, filenames=std::vector of length 1, capacity 1 = {...}, paths=paths@entry=0x0, labels=labels@entry=0x0, errs=errs@entry=0x0, createView=<optimized out>)
    at /usr/src/debug/freecad-git/FreeCAD/src/App/Application.cpp:906
#18 0x00007ffff59928b1 in App::Application::openDocument (this=0x55555570c9c0, FileName=<optimized out>, createView=true) at /usr/src/debug/freecad-git/FreeCAD/src/App/Application.cpp:705
#19 0x00007ffff59aa601 in App::Application::sOpenDocument (args=<optimized out>, kwd=<optimized out>) at /usr/src/debug/freecad-git/FreeCAD/src/App/ApplicationPy.cpp:358
#20 0x00007ffff4d9499c in ??? () at /usr/lib/libpython3.13.so.1.0
#21 0x00007ffff4d5d02c in _PyObject_MakeTpCall () at /usr/lib/libpython3.13.so.1.0
#22 0x00007ffff4d75beb in _PyEval_EvalFrameDefault () at /usr/lib/libpython3.13.so.1.0
#23 0x00007ffff4e40ce5 in PyEval_EvalCode () at /usr/lib/libpython3.13.so.1.0
#24 0x00007ffff4e62870 in ??? () at /usr/lib/libpython3.13.so.1.0
#25 0x00007ffff4e5cc7a in ??? () at /usr/lib/libpython3.13.so.1.0
#26 0x00007ffff4e5068e in ??? () at /usr/lib/libpython3.13.so.1.0
#27 0x00007ffff53179ee in Base::InterpreterSingleton::runString[abi:cxx11](char const*) (this=<optimized out>, sCmd=sCmd@entry=0x555556380fb0 "FreeCAD.openDocument('/home/jffmichi/sketcher_crash.FCStd')")
    at /usr/src/debug/freecad-git/FreeCAD/src/Base/Interpreter.cpp:249
#28 0x00007ffff61ae587 in Gui::Command::_runCommand
    (file=file@entry=0x7ffff67249f0 "/usr/src/debug/freecad-git/FreeCAD/src/Gui/Application.cpp", line=line@entry=650, eType=eType@entry=Gui::Command::App, sCmd=0x555556380fb0 "FreeCAD.openDocument('/home/jffmichi/sketcher_crash.FCStd')")
    at /usr/src/debug/freecad-git/FreeCAD/src/Gui/Command.cpp:725
[... cut ...]
#61 0x00007ffff60cb5ec in (anonymous namespace)::tryRunEventLoop (mainApp=...) at /usr/src/debug/freecad-git/FreeCAD/src/Gui/Application.cpp:2206
#62 (anonymous namespace)::runEventLoop (mainApp=...) at /usr/src/debug/freecad-git/FreeCAD/src/Gui/Application.cpp:2235
#63 Gui::Application::runApplication () at /usr/src/debug/freecad-git/FreeCAD/src/Gui/Application.cpp:2315
#64 0x0000555555559166 in main (argc=<optimized out>, argv=0x7fffffffdc88) at /usr/src/debug/freecad-git/FreeCAD/src/Main/MainGui.cpp:298
```

jffmichi, 2025-01-21:
Here is the build command for GCS.cpp. Notice `_GLIBCXX_ASSERTIONS` being defined which enables bounds checking for `std::vector`.

```
/usr/bin/c++ -DBOOST_DATE_TIME_DYN_LINK -DBOOST_DATE_TIME_NO_LIB -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_FILESYSTEM_NO_LIB -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_PROGRAM_OPTIONS_NO_LIB -DBOOST_REGEX_DYN_LINK -DBOOST_REGEX_NO_LIB -DBOOST_SYSTEM_DYN_LINK -DBOOST_SYSTEM_NO_LIB -DBOOST_THREAD_DYN_LINK -DBOOST_THREAD_NO_LIB -DFMT_SHARED -DHAVE_CONFIG_H -DHAVE_FREEIMAGE -DHAVE_FREETYPE -DHAVE_OPENGL_EXT -DHAVE_RAPIDJSON -DHAVE_TK -DHAVE_XLIB -DOCC_CONVERT_SIGNALS -DPYCXX_6_2_COMPATIBILITY -DQT_CONCURRENT_LIB -DQT_CORE_LIB -DQT_NO_DEBUG -DQT_NO_KEYWORDS -DQT_XML_LIB -DSketcher_EXPORTS -D_OCC64 -I/home/jffmichi/.cache/pacaur/freecad-git/src/build/src/Mod/Sketcher/App/Sketcher_autogen/include -I/home/jffmichi/.cache/pacaur/freecad-git/src/build -I/home/jffmichi/.cache/pacaur/freecad-git/src/build/src -I/home/jffmichi/.cache/pacaur/freecad-git/src/FreeCAD/src -I/home/jffmichi/.cache/pacaur/freecad-git/src/FreeCAD/src/Mod/Sketcher/App -I/home/jffmichi/.cache/pacaur/freecad-git/src/build/src/Mod/Sketcher/App -I/home/jffmichi/.cache/pacaur/freecad-git/src/FreeCAD/src/Mod/Sketcher/App/.. -I/home/jffmichi/.cache/pacaur/freecad-git/src/build/src/Mod/Sketcher/App/.. -I/usr/include/python3.13 -I/usr/include/eigen3 -isystem /usr/include/opencascade -isystem /usr/include/qt6/QtCore -isystem /usr/include/qt6 -isystem /usr/lib/qt6/mkspecs/linux-g++ -isystem /usr/include/qt6/QtXml -isystem /usr/include/qt6/QtConcurrent -Wall -Wextra -Wno-write-strings -march=x86-64 -mtune=generic -O2 -pipe -fno-plt -fexceptions         -Wp,-D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security         -fstack-clash-protection -fcf-protection -Wp,-D_GLIBCXX_ASSERTIONS -g -ffile-prefix-map=/home/jffmichi/.cache/pacaur/freecad-git/src=/usr/src/debug/freecad-git -flto=auto -ffat-lto-objects -fPIC -w -O3 -DNDEBUG -std=gnu++17 -fPIC -I/usr/include -mno-direct-extern-access -MD -MT src/Mod/Sketcher/App/CMakeFiles/Sketcher.dir/planegcs/GCS.cpp.o -MF src/Mod/Sketcher/App/CMakeFiles/Sketcher.dir/planegcs/GCS.cpp.o.d -o src/Mod/Sketcher/App/CMakeFiles/Sketcher.dir/planegcs/GCS.cpp.o -c /home/jffmichi/.cache/pacaur/freecad-git/src/FreeCAD/src/Mod/Sketcher/App/planegcs/GCS.cpp
```

xtemp09, 2025-01-22:
@jffmichi, could you upload your cmake configuration? In the command you've posted, some options usually are absent, such as `-Wp,-D_GLIBCXX_ASSERTIONS`, `-Wp,-D_FORTIFY_SOURCE=2`, `-fstack-clash-protection` or `-fcf-protection`. Where do they come from?

jffmichi, 2025-01-22:
They probably come from my `CFLAGS` and `CXXFLAGS` defined in makepkg.conf:

```
CFLAGS="-march=x86-64 -mtune=generic -O2 -pipe -fno-plt -fexceptions \
        -Wp,-D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security \
        -fstack-clash-protection -fcf-protection"
CXXFLAGS="$CFLAGS -Wp,-D_GLIBCXX_ASSERTIONS"
```

But the problem is the out of bounds access, not the bounds checking.

jffmichi, 2025-01-22:
@AjinkyaDahale might I pull you into this since you made changes to the sketcher code a few commits ago?

AjinkyaDahale, 2025-01-22:
> [@AjinkyaDahale](https://github.com/AjinkyaDahale) might I pull you into this since you made changes to the sketcher code a few commits ago?

Cannot reproduce with:
```
OS: Ubuntu Core 22 (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.1.0dev.39960 (Git) Snap 1275
Build type: Release
Branch: main
Hash: 97d4c952af203e628b6451ef8899b877f067d5c7
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * OpenTheme 2024.9.1
```
or
```
OS: Manjaro Linux (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.1.0dev.39960 +13 (Git)
Build type: Release
Branch: sk-refactor-stage-4
Hash: 495c8beff2e313900af0d5ac526f0915a2630cc1
Python 3.12.7, Qt 6.8.1, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * Curves 0.6.47
  * Curves.backup1662220878.4365516 0.4.4
  * ExtremeProDark 2.5.3
  * ProDarkThemePreferencePack 1.0.0
  * ProDarkThemePreferencePack.backup1662220881.5889003 1.0.0
```

AjinkyaDahale, 2025-01-22:
Looking at the offending lines,
https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Sketcher/App/planegcs/GCS.cpp#L5590-L5602

There are two places this crash can happen `conflictGroups[j - rank]` and `clist[jacobianconstraintmap.at(origCol)]`. The former cannot be out of bounds simply because of definition of the vector and `j`. So the problem has to be in the latter. Hope that helps.

jffmichi, 2025-01-23:
Ok, I recompiled in debug mode with `_GLIBCXX_ASSERTIONS. The line the debugger gave me in release mode seems to have been wrong, sorry for the confusion. In debug mode it crashes on [GCS.cpp (line 5754)](https://github.com/FreeCAD/FreeCAD/blob/ee2f327a964d93239c0cf6fbb8ca813a76123e4b/src/Mod/Sketcher/App/planegcs/GCS.cpp#L5754).

`conflictGroups` is cleared and then used in the loop. Replacing `conflictGroups` with `conflictGroupsOrig` on lines 5754, 5755 and 5759 fixes the issue and seems to adequately represent the functionality before commit a2421d0.

@AjinkyaDahale can you confirm this?

AjinkyaDahale, 2025-01-23:
I can confirm that should be the case. Could you make the PR?

PR #19212 (2025-01-26): Sketcher: fix seg-fault on loading sketch with redundant constraints
fixes #19174
closed 2025-01-26 by chennes