FreeCAD/FreeCAD #19175: BIM: add Element Status workflow

marcuspollio, 2025-01-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

_This is a design issue for a new feature._

The general goal is to design and develop workflows to help in transformation, renovation, restoration and reuse projects.

The first proposition here is to add an Element `Status` concept and property to Objects and some UI tools to manage and take advantage of this new property.
Additionally, a `Phase` concept is added, to group and filter Objects based on their Status property.
Both Status and Phase are simplistic on purpose, i.e. they do not capture the life cycle complexity of a construction project. A better representation could be achieved with Tasks, but this is outside the scope of this proposal.
So it's rather similar to what you find in ArchiCAD ("Renovation filter") and Revit ("Phasing").
For Native-IFC, the Status property is translated with [PropertySet_*element*Common ElementStatus](https://ifc43-docs.standards.buildingsmart.org/IFC/RELEASE/IFC4x3/HTML/lexical/PEnum_ElementStatus.htm).

<details><summary><i>A wall is cut to fit a new window</i></summary>

[image]
</details>

<details><summary><i>A Preference or Tool dialog to manage Status and Phase attributes</i></summary>

[image]
</details>

Status can have a user-defined name, a corresponding IFC ElementStatus and some other rules like Object Style overrides.
Phases can have a user-defined name and a set of rules to map corresponding Status property values.

There would be 4 places in the UI where Status and Phases are exposed to the user :
- A Preference or Tool dialog shown above to define global Status and Phase attributes
- A Phase selection/dropdown in the Status bar to activate a Phase
- A Status selector in the Draft Tray to set the default Status for created objects
- A Status dropdown in the Property view of the selected object

<ins>Questions :</ins>

- The BIM Workbench seems to be the most appropriate context for such workflow and tools. However, it may be useful outside of the AEC field and could be made more generic (using VarSets maybe). What do you think ?
- @tritao Suggested to use Layers instead, where each layer is a step/status, and views have different layers activated or not. Here the idea is to deal with a new Object Property (Status) and avoid copies of similar objects. Do you think Layers already provide enough to facilitate such workflow ?
- Does IFC allow to store user-defined attributes that may not be included in the standard schema, i.e. Phases in this case ?
- I'm willing to implement this, but would need guidance and mentoring, as I'm a code beginner. Does it seem realistic and who would be willing to help ?

### Full version info

```shell
OS: GNU/Linux Ubuntu 22.04 LTS (DE:GNOME/pop/xcb)
Architecture: x86_64
Version: 1.1.0dev.39896 (Git)
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.8.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-22:
@FreeCAD/cad-advisory-group FYI

semhustej, 2025-01-22:
FYI a video showing phasing process in Bonsai. It is 1 year old, I don't know how current it is.

https://youtu.be/_hADRIo-ma4?si=iXqZTwQ3CKXZkgi3

marcuspollio, 2025-01-24:
Hi all !

Thanks for your enthusiasm, ideas and tips on how to get started.
I'll try to add the Status property soon. Will let you know when I have some working code or some more help =P

Then, for the Phase bit, it needs some more thinking. Yes, in a sense, it's quite similar to Layers, but it is not really a structuring/organizational object, more a workflow/UI "helper". It should not change the "real" model objects data, only provide some convenience in the design process and eventually help for the output (TechDraw).

I'll see if Roy's PR https://github.com/FreeCAD/FreeCAD/pull/19207 can help define more clearly how to proceed.

paullee0, 2025-02-03:
Interesting proposal, would like to see how it should develop, work with IFC / Bonsai etc.  :)

marcuspollio, 2025-02-03:
Hi @maxwxyz and @luzpaz, could you please assign me to this issue ? TIA

marcuspollio, 2025-02-04:
@maxwxyz 
Oh, sorry, I forgot to ask for the `1.1` milestone. Do you mind adding that as well ? TIA =P

yorikvanhavre, 2025-02-04:
Done!

marcuspollio, 2025-03-08:
Hi @maxwxyz 
Could you please change the issue type to "task". TIA !