FreeCAD/FreeCAD #19177: Incomplete Vector type for Python for FreeCAD.Base for 3D objects.

DaniloMitrovic, 2025-01-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

`FreeCAD.Base.Vector` is incomplete and don't follow Python initialization ( `__new__` doesn't call  `__init__`). This in turn makes all 3d objects that use it non-inheritable and non usable (such as Axis, Placement ....) . Additional case where this is found is in quaternions (FreeCAD.Base.Rotation, FreeCAD.Base.Matrix), which however isn't the problem with `FreeCAD.Base.Vector2d`.

To reproduce this issue you can try:

```
class Temp(FreeCAD.Base.Vector):pass; # inherits all methods from parent class
 
a = FreeCAD.Base.Vector(1,2,3); # Vector (1,2,3)
b = Temp(1,2,3); # Vector (0,0,0)

class Temp2(FreeCAD.Base.Rotation):pass;  # inherits all methods from parent class

r = FreeCAD.Base.Rotation(1,2,3,4): # Rotation (0.18257418583505536, 0.3651483716701107, 0.5477225575051661, 0.7302967433402214)
q = Temp2(1,2,3,4): # Rotation (0.0, 0.0, 0.0, 1.0)

class Temp3(FreeCAD.Base.Vector2d):pass;
c = FreeCAD.Base.Vector2d(1,2); # Vector2 (1.0,2.0);
d = Temp3(1,2); # Vector2 (1.0,2.0);
```

One of the workarounds I've found is to redefine `__new__` and call `__init__` form there,  but that doesn't initialize class one made (in example `Temp` ) but gets instance of `FreeCAD.Base.Vector` with different name that fails `isinstance` check. This means that casting from Python Primitives (such as complex) is impossible.  

### Full version info

```shell
[code]
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * NewNodes
  * Nodes 0.1.36
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-01-28:
@DaniloMitrovic latest stable is v1.0 way way ahead of v0.21.2.x  
Can you reproduce this issue on v1.0 ? If so please let us know and we'll re-open the ticket.

closed 2025-01-28 by luzpaz

DaniloMitrovic, 2025-03-03:
Yap, it took me a while to install daily version that i can extract python module FreeCAD (failed) so i just put in macro. Issue still persists in 1.1dev - @luzpaz. Here are new info :

```
OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40303 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 145af5cddc9774f00fd4dfb3bd35e94efc5c70ca
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/143.907
Installed mods: 
  * Nodes 0.1.36
```

With python macro of:

```
from FreeCAD import Vector;

defVect = Vector(3,4,1); # default use
print(defVect);

class temp(Vector):pass; # just basic inheritance test

tmpVect = temp(3,4,1);
print(tmpVect);

class temp1(Vector):  # regular init 
    def __init__(self,X,Y,Z):
        super().__init__(X,Y,Z);

tmp1Vect = temp1(3,4,1);
print(tmp1Vect);

class temp2(Vector): # works 
        
    def __new__(self,X,Y,Z):
        cls = super().__new__(self,X,Y,Z);
        cls.__init__(X,Y,Z);
        return cls;
        
    def __init__(self,X,Y,Z):
            self.X = X;
            self.Y = Y;
            self.Z = Z;

tmp2Vect = temp2(3,4,1);
print(tmp2Vect);
print( isinstance(tmp2Vect, temp2));
print( isinstance(tmp2Vect, Vector));
```

And console output: 

```
14:06:45  Vector (3.0, 4.0, 1.0)
14:06:45  Vector (0.0, 0.0, 0.0)
14:06:45  Vector (0.0, 0.0, 0.0)
14:06:45  Vector (3.0, 4.0, 1.0)
14:06:45  False
14:06:45  True
```