FreeCAD/FreeCAD #19181: Meshes from additive loft have segments with inverted faces

edrosten, 2025-01-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

1. Create sketch with concentric circles (45, 44mm diameter)
2. Create sketch with concentric circles (D, D-1 mm diameter), 10mm Z axis offset.
3. Create an additive loft from 1 to 2.

If D is 35, this produces an odd shape with extra segments, and when exported to STL, the faces are inverted. See `repro.zip/*broken*`

If D is 45, the result is as expected. If you then decrease D in steps of 1mm from 45 to 35 the resulting shape is also as expected. See `repro.zip/*OK*`.

[repro.zip](https://github.com/user-attachments/files/18494591/repro.zip)

### Full version info

```shell
OS: Linux Mint 21.3 (custom/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-22:
With what settings and mesher do you mesh the STL? How do you export it?

tarman3, 2025-01-22:
Problem with model you created
Faces is crossover

In both sketches you must keep order
If in first sketch you first draw outside circle, in second  sketch you mus to the same

[image]

tarman3, 2025-01-22:
https://github.com/user-attachments/assets/fb164cce-37c1-4206-9e92-5bbeb3e03933

edrosten, 2025-01-22:
Ah that explains the behaviour I was seeing where changing the size in small steps worked: the constraint solver must have flipped the order of the two circles when I changed the outside circle by a step that was too large.

closed 2025-01-22 by maxwxyz