FreeCAD/FreeCAD #19183: Loft between faces no longer works

macdroid53, 2025-01-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

As be seen in the attached video between 4:30 and 5:08, in past versions the loft performed was possible.

[https://www.youtube.com/watch?v=e6GQJG_T2ew&list=PLWuyJLVUNtc0UszswD0oD5q4VeWTrK7JC&index=42](url)

In 1.1.0.dev it fails with "AdditiveLoft: Loft: At least one section is needed" error.

[LoftFail.FCStd.zip](https://github.com/user-attachments/files/18495627/LoftFail.FCStd.zip)

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39861 (Git)
Build type: Release
Branch: main
Hash: 4cb01753ed532263dddae60e40e8641df56f9185
Python 3.12.7, Qt 6.8.1, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Installed mods: 
  * Silk 0.1.5
  * Defeaturing 1.2.3
  * Curves 0.6.53
  * CurvedShapes 1.0.13
  * DynamicData 2.70.0
  * lattice2 1.0.0
  * sheetmetal 0.7.3
  * Manipulator 1.5.7
  * kicadStepUpMod 11.5.2
  * fasteners 0.5.35
  * OpenTheme 2024.9.1
  * ProDarkThemePreferencePack 1.0.0
  * freecad.gears 1.3.0
  * ExtremeProDark 2.9.0
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-01-21:
The best I can narrow the regression down to, is the last working build I have is:

```
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37302 (Git) AppImage (mid May 2024)
Build type: Release
Branch: main
Hash: 0e24e121eb3e5708c380596e0a9fd583befac977
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * sheetmetal 0.7.3
  * Behave-Dark-Colors 0.1.1
  * fasteners 0.5.35
```

the first failing build I have is:

```
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38043 (Git) AppImage (early July 2024)
Build type: Release
Branch: main
Hash: 561e521817034ea7660de987d8ab7183ad583595
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * sheetmetal 0.7.3
  * Behave-Dark-Colors 0.1.1
  * fasteners 0.5.35

```

Syres916, 2025-01-25:
The first thing I'll do is write a test that passes using 0.21.x because it's quite obvious there's no existing test for this scenario. On Monday I'll see if it was the enabling of TNP that triggered the regression and if so, try to understand where the new code diverts from the old code.

Syres916, 2025-01-27:
Test written: https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:Loft_Two_Faces_dev

Syres916, 2025-01-28:
This is going to take quite a bit of work, the code looks like it for a completely different operation it's changed that much and I don't know whether it's a coincidence but lofting between two faces doesn't work using LinkStable either.

macdroid53, 2025-03-15:
Just a follow up note.
If the selected faces are flat, it succeeds.

Curved faces fail now, but worked previously.

tringenbach, 2025-03-21:
I tried out the unit test, and I guess you already noticed, but it does output:

```
Test issue #19183: Loft tool "Loft between faces no longer works ... AdditiveLoft: Wires are not coplanar.
```

I set a breakpoint on that line where it throws:

```
(lldb) break set -y src/Mod/Part/App/FaceMakerBullseye.cpp:92
Breakpoint 1: no locations (pending).
WARNING:  Unable to resolve breakpoint to any actual locations.
(lldb) run -c -t TestPartDesignApp.TestLoft.testTwoFacesAdditiveLoftCase
```

Not sure if this is helpful but here's the backtrace and a few variables. I'm not entirely sure, but it seems like it only gets passed one Wire from here:

```c++
        TopoShape back;
        if (wiresections[0].back().shapeType() != TopAbs_VERTEX) {
            std::vector<TopoShape> backwires;
            for(auto& sectionWires : wiresections)
                backwires.push_back(sectionWires.back());
            back = TopoShape(0).makeElementFace(backwires);
        }
```

backtrace:
```
Process 55522 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = breakpoint 1.1
    frame #0: 0x000000010ddf7531 Part.so`Part::FaceMakerBullseye::Build_Essence(this=0x00007fd44b9a7ba0) at FaceMakerBullseye.cpp:92:13 [opt]
   89  	        }
   90  	        BRepLib_FindSurface planeFinder(comp, -1, /*OnlyPlane=*/Standard_True);
   91  	        if (!planeFinder.Found())
-> 92  	            throw Base::ValueError("Wires are not coplanar.");
   93  	        plane = GeomAdaptor_Surface(planeFinder.Surface()).Plane();
   94  	    }
   95
Target 0: (FreeCAD) stopped.
warning: Part.so was compiled with optimization - stepping may behave oddly; variables may not be available.
(lldb) bt
* thread #1, queue = 'com.apple.main-thread', stop reason = breakpoint 1.1
  * frame #0: 0x000000010ddf7531 Part.so`Part::FaceMakerBullseye::Build_Essence(this=0x00007fd44b9a7ba0) at FaceMakerBullseye.cpp:92:13 [opt]
    frame #1: 0x000000010ddf08ee Part.so`Part::FaceMaker::Build(this=0x00007fd44b9a7ba0, (null)=<unavailable>) at FaceMaker.cpp:128:11 [opt]
    frame #2: 0x000000010dda235b Part.so`Part::TopoShape::makeElementFace(this=0x00007ff7bfefa750, shapes=<unavailable>, op=0x0000000000000000, maker=<unavailable>, plane=<unavailable>) at TopoShapeExpansion.cpp:4605:13 [opt]
    frame #3: 0x0000000112c48772 _PartDesign.so`PartDesign::Loft::execute(this=0x00007fd44c241400) at FeatureLoft.cpp:189:33 [opt]
    frame #4: 0x000000010095c1cd libFreeCADApp.dylib`App::DocumentObject::recompute(this=0x00007fd44c241400) at DocumentObject.cpp:172:22 [opt]
    frame #5: 0x000000010db0cecc Part.so`Part::Feature::recompute(this=<unavailable>) at PartFeature.cpp:107:33 [opt]
    frame #6: 0x0000000112baad02 _PartDesign.so`PartDesign::Feature::recompute(this=0x00007fd44c241400) at Feature.cpp:82:31 [opt]
    frame #7: 0x00000001008d4476 libFreeCADApp.dylib`App::Document::_recomputeFeature(this=0x00007fd44b0c5800, Feat=0x00007fd44c241400) at Document.cpp:3348:32 [opt]
    frame #8: 0x00000001008d2181 libFreeCADApp.dylib`App::Document::recompute(this=0x00007fd44b0c5800, objs=<unavailable>, force=<unavailable>, hasError=0x0000000000000000, options=<unavailable>) at Document.cpp:3058:31 [opt]
    frame #9: 0x00000001009d1613 libFreeCADApp.dylib`App::DocumentPy::recompute(this=0x00006000015e5dc0, args=<unavailable>) at DocumentPyImp.cpp:701:31 [opt]
    frame #10: 0x00000001009ca280 libFreeCADApp.dylib`App::DocumentPy::staticCallback_recompute(self=<unavailable>, args=<unavailable>) at DocumentPy.cpp:1989:57 [opt]
    frame #11: 0x000000010184b634 libpython3.13.dylib`cfunction_call + 100
    frame #12: 0x000000010197c70d libpython3.13.dylib`_PyEval_EvalFrameDefault + 69181
    frame #13: 0x00000001017c7f1e libpython3.13.dylib`method_vectorcall + 206
    frame #14: 0x000000010197fb3d libpython3.13.dylib`_PyEval_EvalFrameDefault + 82541
    frame #15: 0x00000001017c4ac0 libpython3.13.dylib`_PyObject_Call_Prepend + 176
    frame #16: 0x0000000101890d23 libpython3.13.dylib`slot_tp_call + 211
    frame #17: 0x000000010197c70d libpython3.13.dylib`_PyEval_EvalFrameDefault + 69181
    frame #18: 0x00000001017c7f1e libpython3.13.dylib`method_vectorcall + 206
    frame #19: 0x000000010197fb3d libpython3.13.dylib`_PyEval_EvalFrameDefault + 82541
    frame #20: 0x00000001017c4ac0 libpython3.13.dylib`_PyObject_Call_Prepend + 176
    frame #21: 0x0000000101890d23 libpython3.13.dylib`slot_tp_call + 211
    frame #22: 0x000000010197c70d libpython3.13.dylib`_PyEval_EvalFrameDefault + 69181
    frame #23: 0x000000010196917d libpython3.13.dylib`PyEval_EvalCode + 157
    frame #24: 0x0000000101a5be80 libpython3.13.dylib`run_eval_code_obj + 144
    frame #25: 0x0000000101a5b7be libpython3.13.dylib`run_mod + 142
    frame #26: 0x0000000101a5df00 libpython3.13.dylib`PyRun_StringFlags + 112
[... 180 lines cut ...]
        }
      }
    }
    myOrient = TopAbs_FORWARD
  }
}
(lldb) p myWires
(std::vector<TopoDS_Wire>) size=1 {
  [0] = {
    TopoDS_Shape = {
      myTShape = {
        entity = 0x0000600001554700
      }
      myLocation = {
        myItems = {
          myNode = {
            entity = 0x0000600007192bc0{...}
          }
        }
      }
      myOrient = TopAbs_REVERSED
    }
  }
}
```

PR #22165 (2025-08-04): PD: Fix loft between curved faces
Inspired by a9100c1425

Fixes #19183

Also reported in the forum:
https://forum.freecad.org/viewtopic.php?p=806495
https://forum.freecad.org/viewtopic.php?t=88234
    
The actual reason for the failure is that the new code uses `FaceMakerBullseye` method which expects coplanar wires.
The code was using `FaceMakerCheese` previously which changed in LinkStage with their commit [PD: use FaceMakerBullseye for feature Loft](https://github.com/realthunder/FreeCAD/commit/93ce3edfe7ff714d24fdfea3a0ae1dde10714f19) and was imported here with TNP mitigations.
    
As cha [... cut]
closed 2025-08-04 by hyarion