FreeCAD/FreeCAD #19184: Part Design and Part Thickness fails for a very simple revolution feature around Z axis, but not for X or Y axes

kwahoo2, 2025-01-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Part Design Thickness fails for a very simple shape, a hemisphere made by Revolution around Z axis. But not for the very similar ones created by revolution around Y or X.

To reproduce:
1. Download (change extension to FCstd):
[Hemisphere Z - error.zip](https://github.com/user-attachments/files/18495534/Hemisphere.Z.-.error.zip)
2. Select the flat face and apply Part Desgn Thickness.

It will fail with:
```20:22:18  <PartDesign> FeatureThickness.cpp(150): Exception on making thick solid: BRep_API: command not done```
```20:22:18  Thickness: Failed to make thick solid```

3. Change to Revolution base axis from Z to X.  Select the flat face. Apply thickness.
4. A Thickness feature will be successfully created.

Part thickness fails too.
But not if you create a copy:
1. Use Part-Create a copy-Create simple copy
2. Select the flat face on the copy
3. Apply the Part Thickness
4. A Thickness shape will be created without error.

### Full version info

```shell
OS: Fedora Linux 41 (KDE Plasma) (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * OpticsWorkbench 1.0.26
  * fasteners 0.5.35
  * sheetmetal 0.6.13
  * Curves 0.6.53
  * freecad.gears 1.3.0
  * freecad-xr-workbench 0.9.3
  * kicadStepUpMod 11.4.4
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-27:
@FlachyJoe FYI

berkakinci, 2025-04-14:
I thought this may be related.
In the attached overly-simplified case, I don't get the `Thickness: Failed to make thick solid` error.
However, geometry check fails after Thickness.  Note there are two working, one failing part in this file.

This is a simplified version of what I reproduced in real use.  In that case, trying to add a pocket to the broken geometry fails.
It also fails (sometimes silently) in this simplified example too.

[BrokenThicknessAfterDraft.FCStd.zip](https://github.com/user-attachments/files/19741748/BrokenThicknessAfterDraft.FCStd.zip)

berkakinci, 2025-09-24:
Reproduced on latest stable.  (Sorry; I don't have any dev version installed.)
```
OS: macOS 15.7
Architecture: arm64
Version: 1.0.2.39319 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.2)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.11.13, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * kicadStepUpMod.backup1753993021.644542 (Disabled)
  * kicadStepUpMod.backup1753993062.0308852 (Disabled)
  * kicadStepUpMod
```

luzpaz, 2025-09-24:
@berkakinci any chance you can test on the 1.1 dev build ? https://github.com/FreeCAD/FreeCAD/releases

Backup any 1.0.x models before opening them in 1.1 if you do.