FreeCAD/FreeCAD #19189: Trimex command swaps start and end of line

kpemartin, 2025-01-22:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When you use the Trimex command to move one end of a Draft Line, the Start and End points may be interchanged. That is, the point that is moved by the command becomes the Start and the point not moved becomes the End, regardless of which was originally the Start and End.

This is normally not a problem, but if there is a parametric reference to the Start or End of the line, the reference can end up referring to the wrong end of the line.

To reproduce, in Draft workbench, on a new document:
Draw a line from the origin to (2,0,0). Its Name and Label will both be Line
Draw another line from the origin to (0,1,0). Its Name and Label will both be Line001

Here is the Top view with nothing selected:
[image]

The Start of both lines will be the origin.
Now link the second line to the first by setting Line001.Start to be Line.Start.x, Line.Start.y, Line.Start.z. There may be a faster way of doing this but it can be done in the Properties of Line001 by opening up and modifying the individual coordinates of its Start.

[image]

Now use Trimex to move the Start of Line (at 0,0), for instance drag it to the left a bit. The Start of Line001 follows along:

[image]

Now use Trimex to move the End of Line (at 2,0), for instance drag it to the right a bit. The start of Line001 jumps to the new right end of Line because this point is now its Start.

[image]

[image]

The Trimex command should be ensuring that the line ends maintain their identities as to which is the Start and which is the End.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39935 (Git)
Build type: Release
Branch: master
Hash: 49b304ae558553f9acd5872ccd30673bbd60bd19
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.0
Locale: English/Canada (en_CA)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Classic/
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-22:
@Roy-043 FYI

semhustej, 2025-02-14:
Here is an example of a problem with Arch_Wall alignment this might create in the BIM workbench:

https://github.com/user-attachments/assets/d9a9d1da-028f-4b09-9e0a-5be243a76d41

maxwxyz, 2025-02-14:
@yorikvanhavre FYI - I hope this won't happen with the new wall tool :)

semhustej, 2025-02-20:
> FYI: With [Draft Edit](https://wiki.freecad.org/Draft_Edit) you can reverse the direction of a line/wire.

This is a neat trick, thank you.

DesignWeaver3D, 2026-10-11:
I confirmed the bug is still present in v26.3rc1 and will submit a PR with a fix soon.

While testing I found a separate problem. While each use of Trimex registers a transaction in the Undo list, undoing each one doesn't restore the line without a manual recompute. The cause is a lack of recompute before the transaction closes. I'll be submitting a separate PR to add a recompute prior to commitTransaction() in Trimex.trimObject().

semhustej, 2026-10-11:
> While testing I found a separate problem. While each use of Trimex registers a transaction in the Undo list, undoing each one doesn't restore the line without a manual recompute. The cause is a lack [...]

I can confirm this behavior, it just did not occur to me to report it as I took it as a fact. Thanks for working on this.

DesignWeaver3D, 2026-10-11:
Draft: fix Undo not restoring the shape after Trimex- #33458

closed 2026-10-16 by Roy-043