FreeCAD/FreeCAD #19190: Check box for active body/part.

oficinerobotica, 2025-01-22:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

On moderately complex projects I often find myself needing to activate a particular part or body. Having a check box for active part/body in the tree view alongside the eye icon for the active element would greatly enhance the experience , making the operation allot faster and also giving a clear visual cue about what particular element is active.  This is especially true when modeling "in context" in the new assembly workbench .

I know that double clicking on a element tree activates it but from real work experience I can say that I find my self more often deactivating elements instead of activating them by double click. 
On a side note, for some reason, the open theme doesn't give a clear visual cue about the active element other than making the label bold but that can be easily missed in a crowded tree view.

In the screenshot the active part is "spindle" and the active body is "spindle body" , it would be impossible to guess at a simple glance.

[image]

### Full version info

```shell
OS: KDE neon 6.2 (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39896 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * fasteners 0.5.35
  * PieMenu 1.11.2
  * OpenTheme 2024.9.1
  * freecad.gears 1.3.0
  * sheetmetal.backup1729482375.2960367 0.5.1 (Disabled)
  * sheetmetal 0.7.3
  * Curves 0.6.53
  * Gridfinity 0.7.2
  * ArchTextures
  * Assembly4 0.50.15
  * CurvedShapes 1.0.13
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

kaiwas, 2025-01-22:
A similar problem has already been opened. And there is even a workaround.
https://github.com/FreeCAD/FreeCAD/issues/17773

But I agree that there is a lack of emphasis on the active body now. I noticed this even without seriously using the PD workbench.

maxwxyz, 2025-01-22:
@FreeCAD/design-working-group FYI

I'll leave this issue open, #17773 for the indicator / visibility, this one to quickly toggle active state.

pierreporte, 2025-01-22:
There can already be checkboxes in the tree view, right? What is their role?

maxwxyz, 2025-01-22:
Pure selection, similar as in the file explorer to select multiple objects

pierreporte, 2025-01-22:
Then I don't think it's a good idea to use a checkbox to make a container active.

The double click is the easiest solution. To make it more precise, what about allowing to double click on the children of the container to activate it? It would be like Catia. To show the active container, making the color of lines outside the active container could be muted. Alibre does a similar thing in assembly when editing a childre, but makes the active container blue.

oficinerobotica, 2025-01-23:
I agree that having checkboxes all over the tree looks crowded but there should really be a way of indicating the active container independent of the theme used. The default theme highlights the active container with a background color of the active tree element , open theme does not. 

Just as reference here's how Fusion does it. The small dot(checkbox) on the right of the tree entry is always visible for the active container. For the inactive elements, the checkbox becomes visible only on hoover over the respective tree entry. This could be an elegant solution.

[image]

maxwxyz, 2025-01-24:
Maybe it could be the same as a new (better) tip indicator, so the UI is consistent: https://github.com/FreeCAD/FreeCAD/issues/13016
@kadet1090 FYI for different tree views and part/part design merge

hyarion, 2025-01-25:
Check boxes wouldn't be good for this as they are not mutually exclusive (only one object can be active.

Radio button would work better.

dzid26, 2025-01-31:
> Just as reference here's how Fusion does it.

This is related to my https://github.com/FreeCAD/FreeCAD/issues/18900 - I was already proposing the  use radio buttons for _feature_ activation.

> Maybe it could be the same as a new (better) tip indicator
Depends what the new tip indicator is. It it's a horizontal line then not really.  If it is an arrow then it could point to the active body/object. Once it is expanded the arrow could go down and point to an actual tip.

hyarion, 2025-02-03:
This sounds more like a discussion than a feature request. How about closing this, continue the discussion elsewhere and open a feature request when it's been solidified?