FreeCAD/FreeCAD #19191: [Sketcher] Two regressions introduced by PR 17312, new geometry vertices not visible without rotating sketch and flickering of axes on opening of new sketch

Syres916, 2025-01-22:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Having spent a while trying to ensure that a graphics card driver had not caused these issues, I ran multiple Git Bisects back in monthly steps to find when I last could create a new sketch and new geometry with smooth axes movement and the new geometry having immediately visible vertices. The conclusion was https://github.com/FreeCAD/FreeCAD/commit/5969df37f4bd278798c13e2910acb8b96965a4a2 introduced both issues.

New rectangle:

[image]

Same rectangle after rotating the sketch slightly:

[image]

Fortunately this wasn't backported so the latest 1.0 branch isn't affected.

As far as the flickering axes, IMHO photo sensitive users should have the ability to at the very least to use the old axes length method via a checkbox in Preferences.

### Full version info

```shell
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.39314 (Git)
Build type: Release
Branch: (no branch, bisect started on main)
Hash: 5969df37f4bd278798c13e2910acb8b96965a4a2
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * sheetmetal 0.7.3
  * Behave-Dark-Colors 0.1.1
  * fasteners 0.5.35
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-01-22:
@FlachyJoe I'll look into the axes UI Preferences change, could you please look at the vertices not showing immediately?

maxwxyz, 2025-01-22:
@Syres916 I cannot reproduce. Could you share a file? Tried sketches on all base planes, with and without grid and show sketch on/off (clipping)

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39896 (Git) Conda
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * Curves 0.6.51
  * fasteners 0.5.32
  * freecad.gears 1.2.0
  * Manipulator 1.5.7

```

Syres916, 2025-01-22:
@maxwxyz I'm not overly surprised that this only applies to certain users as I would have expected it to be reported way before now, it probably depends on Coin versions and even Qt versions combined. There's no point in supplying a file as when I open an existing file it isn't an issue so the simple procedure is:

```
Create New Document
Switch to the Sketcher Wb
Create New Sketch and leave radio button on X-Y Plane and click OK
Select the default Rectangle tool and create a rectangle of any size crossing the origin as screenshot above.
```

Even running a macro created from the Python Console output of the failed rectangle doesn't recreate the issue as it needs a Zoom To Fit which suddenly brings the vertices to life so it looks to me as though there needs an extra _re-paint_ needed.

Personally I would say the Blocker tag is not needed here.

Syres916, 2025-01-22:
I was planning on adding a checkbox to Preferences > Sketcher > Display like (ignore the font and color):

[image]

@FlachyJoe probably best to leave the whole change to me as I've tried isolating each function and had to revert the entire commit to get it to behave.

Syres916, 2025-01-22:
Wait on CI tests of https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:Allow_use_legacy_axes_Sketcher_dev

maxwxyz, 2025-01-22:
@Syres916 do we need another preference for this? Can anyone else reproduce this issue or will this be a preference just for one Coin version + system? Can it be dependent on the Coin version?
@FreeCAD/design-working-group what is your opinion?

Syres916, 2025-01-22:
> [@Syres916](https://github.com/Syres916) do we need another preference for this? Can anyone else reproduce this issue or will this be a preference just for one Coin version + system? Can it be dependent on the Coin version? @FreeCAD/design-working-group what is your opinion?

Rather than repeat myself regarding Coin version etc see https://github.com/FreeCAD/FreeCAD/pull/19195#issuecomment-2608079809

hyarion, 2025-01-23:
(I haven't gone through the infinite axis code in detail yet so the following is only based on what I see and remember about how sketcher renders the view.)

To me it looks like the camera is over adjusting znear and/or zfar in the projection matrix so it gets too tight. That would also explain the missing geometries as those have a small offset to fix render order.

Is this happening if you have any other geometry in the viewport?

maxwxyz, 2025-01-23:
@Rexbas FYI

hyarion, 2025-01-24:
@Syres916 it sounds like this could be related to the bounding boxes of the geometry.
Could you try this PR https://github.com/FreeCAD/FreeCAD/pull/19218 and share a screen recording of the values when creating a new sketch in a new document as before?

Syres916, 2025-01-24:
@hyarion I hope this has the info you require in it 

https://github.com/user-attachments/assets/0a7872a9-2c64-41d1-85dc-4c476724992e

Rexbas, 2025-07-01:
@Syres916, how reproducible is this for you with the latest build? I cannot reproduce this issue. But I guess it could happen when the near and far distance of the camera are too close too each other, which is what happens in an empty scene. At least that is what is happening in the recent issues about the point not being visible.

Rexbas, 2025-08-01:
@WesleyAC, can you try it using the weekly build?

FEA-eng, 2025-10-22:
@Syres916 This seems to be OS-specific. Can you confirm if it's still relevant?