FreeCAD/FreeCAD #19194: Base axis changed in version 1.1.0dev.39896

tomunger, 2025-01-22:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

From version 1.1.0dev.39747 (12/27) to version 1.1.0dev.39896 (1/17) the alignment of objects in part design changes. I create a file in version 1.1.0dev.39747 with a sketch on the XY plane. When I open it in version 1.1.0dev.39896, the sketch is now in the YZ plane.

Sketch in version 1.1.0dev.39747 on XY plane.  

[image]

Sketch in version 1.1.0dev.39896 now on XZ plane

[image]

**

**: When I open this file in 1.0 it is aligned correctly.  If I use 1.0 to edit and save, it is aligned correctly in  1.1.0dev.39747.  So, all good between those versions.

The few files that I have opened in the latest version look OK, but alignment is changed and the effect of edits is incomprehensible.  

**Expected behavior**:  an object attached to the XY plane should remain attached to the XY plane. 

**Reproduce**:  Open this file in a previous version and the sketch is on the XY plane.  Open it in 1.1.0dev.39896 and the sketch is on the XZ plane

[axis_problem.FCStd.zip](https://github.com/user-attachments/files/18507111/axis_problem.FCStd.zip)

### Full version info

```shell
OS: macOS 15.1.1
Architecture: arm64
Version: 1.1.0dev.39896 (Git) Conda
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * DynamicData 2.70.0
  * Manipulator 1.5.0
  * fasteners 0.5.21
  * lattice2 1.0.0
  * CurvedShapes 1.0.4
  * Curves 0.6.13
  * Assembly3 0.12.0
```

### Subproject(s) affected?

None

### Anything else?

I asked about this on the [Help Using FreeCAD forum](https://forum.freecad.org/viewtopic.php?p=805810#p805810) and was told there was some change in the base coordinates for more correctness.  I searched for and read some issues related to that but I don't really understand what is being done.  However, it can't be that the intention is that the alignment of every part design object changes.  

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

furgo16, 2025-01-22:
Is this related to https://github.com/FreeCAD/FreeCAD/issues/19060?

tomunger, 2025-01-22:
I read that one.  Seems that it could be related.  But the issue I'm reporting occurs for the most basic Part Design object.  It made the few files I opened un-usable.  That is, I edited a sketch and it resulted in a change that I could not comprehend.  I could provide that example, but I'm fairly sure it relates to the simple example provided above.

maxwxyz, 2025-01-23:
Might be because of the migration for the datums @wwmayer @PaddleStroke
If only a dev version is affected, I'd close this.

PaddleStroke, 2025-01-23:
@tomunger
Basically with core datums I made a migration to fix the placements of axis. But this modification was reversed as it was causing too many issues. But there's no migration back. See https://github.com/FreeCAD/FreeCAD/issues/19058

maxwxyz, 2025-01-23:
Correct

closed 2025-01-23 by maxwxyz