FreeCAD/FreeCAD #19199: Draft/TechDraw: Linked draft annotations not visible on page in TechDraw_DraftView

semhustej, 2025-01-23:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In the current 1.1 dev version, linked draft annotations (tested with Draft_Text, Draft_Dimension) are not visible in the TechDraw_DraftView.

Sample file with linked dimension and text not visible on page: 

[FC-linked-dimension.FCStd.zip](https://github.com/user-attachments/files/18518496/FC-linked-dimension.FCStd.zip)

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39896 (Git) Conda
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

Draft

### Anything else?

This was working in Version: 1.1.0dev.39380 (Git) Conda , so it seems to me like a regression.

This issue was worked on in the 1.1 release cycle: https://forum.freecad.org/viewtopic.php?p=795189#p795189 . I am not sure when it was broken, but I think it worked on January 10th, as I was using weekly around that time to print out linked dimensions.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-27:
@WandererFan @Roy-043 FYI

Roy-043, 2025-01-28:
Not a regression. AFAICT this is partly a duplicate of #18511. I'll close that one.

semhustej, 2025-01-28:
> Not a regression. AFAICT this is partly a duplicate of [#18511](https://github.com/FreeCAD/FreeCAD/issues/18511). I'll close that one.

I thought it worked, but I guess the dimension were inside of a linked group previously. I am sorry for the confusion.

PR #19296 (2025-01-28): Draft: fix handling of annotation Links in TD DraftViews
Fixes #19199.
commit 830b2aad (2025-01-28): Draft: fix handling of annotation Links in TD DraftViews

Fixes #19199.
closed 2025-01-28 by Roy-043