FreeCAD/FreeCAD #19200: The new CAM simulator can run only once

xtemp09, 2025-01-23:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Since the issue #9403 has been closed and can't be reopened, I started the tests anew to detect memory leaks again. Since _pybind11_ offers less memory leaks, I turned it on.

_FreeCAD_ with the old CAM simulator creates _2 804 218_ bytes if the simulation is run twice.

The new CAM simulator creates _4 103 549_ bytes of memory leaks in the first run. The second time the simulator **does not start**, I don't know why. I closed the project and reopened, it does not start anyway.

Here are the leak reports. The new CAM simulators leaks a little bit.

[leaks-old-CAM.txt](https://github.com/user-attachments/files/18518911/leaks-old-CAM.txt)

[leaks-new-CAM.txt](https://github.com/user-attachments/files/18518913/leaks-new-CAM.txt)

If _FreeCAD_ just launched once, without doing anything, it gives _2 509 052_ bytes of memory leaks.

The test file is [drill_sheet_jig.zip](https://github.com/user-attachments/files/18518984/drill_sheet_jig.zip).

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39970 (Git)
Build type: Debug
Branch: master
Hash: ee2f327a964d93239c0cf6fbb8ca813a76123e4b
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

### Subproject(s) affected?

CAM

### Anything else?

_FreeCAD_ has been compiled with the following configuration.

<details><summary>cmake configuration</summary>

```ruby
cmake ../freecad-source -G Ninja \
-D CMAKE_BUILD_TYPE=Debug \
-D CMAKE_INSTALL_PREFIX=/home/xtemp09/path-test/ \
\
-D CMAKE_C_COMPILER=clang-19 -D CMAKE_CXX_COMPILER=clang++-19             \
-D CMAKE_C_FLAGS_DEBUG="-glldb -Og -fsanitize=leak -fno-omit-frame-pointer"   \
-D CMAKE_CXX_FLAGS_DEBUG="-glldb -Og -fsanitize=leak -fno-omit-frame-pointer" \
-D LINK_FLAGS_DEBUG="-fsanitize=leak -fuse-ld=mold"                       \
-D FREECAD_USE_PYBIND11=On \
\
-D CMAKE_AR=/usr/lib/llvm-19/bin/llvm-ar               \
-D CMAKE_ADDR2LINE=/usr/lib/llvm-19/bin/llvm-addr2line \
-D CMAKE_MT=/usr/lib/llvm-19/bin/llvm-mt               \
-D CMAKE_NM=/usr/lib/llvm-19/bin/llvm-nm               \
-D CMAKE_OBJCOPY=/usr/lib/llvm-19/bin/llvm-objcopy     \
-D CMAKE_OBJDUMP=/usr/lib/llvm-19/bin/llvm-objdump     \
-D CMAKE_RANLIB=/usr/lib/llvm-19/bin/llvm-ranlib       \
-D CMAKE_READELF=/usr/lib/llvm-19/bin/llvm-readelf     \
-D CMAKE_RC_COMPILER=/usr/lib/llvm-19/bin/llvm-rc      \
-D CMAKE_STRIP=/usr/lib/llvm-19/bin/llvm-strip
```

</details>

<details><summary>Version info</summary>

```
$ clang-19 --version
Ubuntu clang version 19.1.7 (++20250114103332+cd708029e0b2-1~exp1~20250114103446.78)
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/lib/llvm-19/bin
```

```
$ mold --version
mold 2.30.0 (compatible with GNU ld)
```

```ruby
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39970 (Git)
Build type: Debug
Branch: master
Hash: ee2f327a964d93239c0cf6fbb8ca813a76123e4b
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

</details>

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

tarman3, 2025-01-23:
Open your file and run new simulation many time without problem

I noticed strange things in your project
1. Objects which must be inside **Job** is outside (Operation, Model, Tools, ...) 
2. Tool shape is strange. It is a hole inside

<details>
  <summary>FreeCAD 1.1.0dev.39896 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39896 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.35; sheetmetal 0.7.3; btl 0.9.9; Silk 0.1.5; Curves 0.6.53; Defeaturing 1.2.3

```

</details>

xtemp09, 2025-01-23:
Thank you for taking a look at this. The project has been created in _0.19.24291 (Git) AppImage_, obtained from here https://forum.freecad.org/viewtopic.php?f=15&t=63656

I launched the new simulator in AppImage and it worked, at least twice. I don't know where the problem comes from, perhaps, _OpenGL_. There is nothing in the logs.

```ruby
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39896 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

tarman3, 2025-01-23:
Did you try to create a new project with **Job** in 1.1.0dev.39896 ?

xtemp09, 2025-01-23:
I haven't tried to recreate the project or a job with newer version, but, for some reason, the new CAM simulator now can be launched more than once. OpenGL seems to be picky when it comes to virtual machines, so let's close the issue for now.

The new CAM simulator leaks not so much. _pybind11_ seems to be better than _Boost.Python_. Here is the leak report for posterity. 

[leaks-new-CAM2.txt](https://github.com/user-attachments/files/18520439/leaks-new-CAM2.txt)

The new CAM simulator gives _4 304 501_ bytes of leaked memory in comparison to the old one (_2 804 218_ bytes).

luzpaz, 2025-01-28:
@xtemp09 can you add the new leak info to #9403 ? TIA

xtemp09, 2025-01-28:
@luzpaz, the leaks are minuscule (4 Mb), in comparison to the ones reported before (28 Mb). If somebody has time, he can take a look at this.

shaise, 2025-01-29:
Can someone give me some pointers how to search for real leak issues in the report?
Which are one-time allocations and which are recurring?

luzpaz, 2025-01-31:
@xtemp09 👆

xtemp09, 2025-02-01:
@luzpaz, my minute level of expertise in this area does not allow to provide answers to these questions. How would I know which leaks are one-time allocations and which are not?

I can give an advice: usage of combination of tools usually provide more information than just one. _valgrind_, leak sanitizer and other means used together can help to narrow down the leaks.