FreeCAD/FreeCAD #19203: Wiki: 1600+ page broke due to bulk edit

Roy-043, 2025-01-23:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Due to a bulk edit almost all translated pages in the Wiki are now broken. This needs to be reverted ASAP. It is blocking almost everything.

@Reqrefusion In the future please test such edits with small batches first.

https://forum.freecad.org/viewtopic.php?t=94134

### Full version info

```shell
Does not apply
```

### Subproject(s) affected?

Project Tools & Websites

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Reqrefusion, 2025-01-23:
Problem solved. This is actually a problem based on something quite funny. I was clearing the commit history of the relevant Repository, but I forgot to turn off mirroring, and this actually revealed a shortcut I had made in the past. On the All pages page, some pages do not have their own version, they only exist as /en, I realized this after downloading them all. To solve this, I transferred the contents of the /en folder to the main folder, deleted the existing ones and left the ones that were not. However, pages with /en lack translation tags. When I cleared the commit history, it saw this difference and then reflected the changed and different thing on the wiki.

Reqrefusion, 2025-01-23:
Also, 1600+ is an exaggeration, there is a change between 150-200, if translations are included, only 800-1000 is affected.

Roy-043, 2025-01-23:
You are right about the wrong number of affected pages. 308 pages are affected, 1600+ pages are not affected (i scrolled down the https://wiki.freecad.org/Special:PageTranslation page and missed a heading). Now all those 308 pages have to be marked for translation...

IMO something seriously went wrong here. We should not make light of this.

closed 2025-01-23 by Roy-043

Reqrefusion, 2025-01-23:
> You are right about the wrong number of affected pages. A little over 300 pages are affected, 1600+ pages are not affected (I scrolled down the https://wiki.freecad.org/Special:PageTranslation page and missed a heading). Now all those 300 pages have to be marked for translation...
> 
> IMO something seriously went wrong here. We should not make light of this.

I don't take it lightly, I think even one page is enough and no page should be affected. It was a problem that occurred to me absentmindedly, but I solved it and fixed it so that it wouldn't happen again. I'm thinking of adding some tools to the wiki for such things in the future. But the problem is not caused by the bot, but by me. Technically, the bot was supposed to change all the pages in the wiki, but since the comparison system worked correctly, it only changed the pages I specified. In other words, it worked correctly and from now on, it will never make any mistakes since all pages are up to date.