FreeCAD/FreeCAD #19204: Std_Measure: save 'Show Delta' state between commands

semhustej, 2025-01-23:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When users sets `Show Delta` option of `Std_Measure` tool to off, the setting is not preserved when the tool is invoked again.

Steps to reproduce:

- start Std_Measure tool
- `Show Delta` is turned on
- select distance to be measured
- disable `Show Delta` by unchecking the box
- close the tool
- start Std_Measure tool again
- `Show Delta` is turned on

Suggested behavior:
`Show Delta` state should be saved between repeated invoking of the Std_Measure command. It is expected, that if the user turns the option off, he/she does not need delta values information. It is also expected, that he/she would know, how to turn this option if the desire for delta information arises.
I think it would be ideal, if the state of `Show Delta` would be saved system wide for current user.

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39896 (Git) Conda
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

harikrishna-au, 2025-02-04:
@semhustej Is this issue still open? If so, can I take a look and work on solving it?

semhustej, 2025-02-04:
> [@semhustej](https://github.com/semhustej) Is this issue still open? If so, can I take a look and work on solving it?

Hi,

yes, it seems it is still not fixed. But let's wait for what @luzpaz has to say as he manages this issue, I just file the issue. 

Thanks for being active in trying to solve the issue :)

closed 2025-02-11 by chennes

semhustej, 2025-02-12:
@harikrishna-au seems like this one has already been taken care of.

PR #19430 (2025-02-11): Measurement: save 'Show Delta' state between commands #19204
When users set Show Delta option to off in the Measure tool, the setting was not being preserved when the tool was invoked again. This was due to     using an uninitialized value as the default when loading the preference.  

Changes staged in: FreeCAD\src\Mod\Measure\Gui\TaskMeasure.cpp 

Fixed by:
- Using explicit 'true' as default value when loading Show Delta preference 
- Maintains existing save functionality
- Ensures consistent behavior on first use 
- Fixes #19204