FreeCAD/FreeCAD #19213: Sketcher + Curves: Control points and circles disappear from bsplines after exiting Sketcher

theosib, 2025-01-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Create a cylinder in Part workbench
2. Select the side
3. Go to Curves workbench
4. Create a sketch on surface
5. In the sketch, create a bspline
6. Exit Sketcher
7. Re-enter Sketcher

Expected behavior:
The control points and circles should still be there.

Actual behavior:
The control points and circles are deleted.

### Full version info

```shell
OS: macOS 15.1.1
Architecture: arm64
Version: 1.1.0dev.39896 (Git) Conda
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US) [ OS: C/Default (C) ]
Stylesheet/Theme/QtStyle: unset/Classic/Qt default
Installed mods: 
  * freecad.gears 1.3.0
  * Pyramids-and-Polyhedrons
  * fasteners 0.5.29
  * Curves 0.6.50
```

### Subproject(s) affected?

Sketcher

### Anything else?

[wavy2.FCStd.zip](https://github.com/user-attachments/files/18529327/wavy2.FCStd.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

theosib, 2025-01-24:
Alternative steps:

1. Create a cylinder in Part workbench
2. Create a sketch with some bsplines
3. Put a construction geometry box around it
4. Exit sketcher
5. Set the side of the cylinder as the attachment support for the sketch

Now at this point, if you exit and re-enter the sketch, the control points will still be there.

6. Select the sketch
7. Go to Curves and slick Sketch on surface

Now if you go into the sketch, you'll find that the control points and circles have disappeared.

maxwxyz, 2025-01-24:
@AjinkyaDahale @tomate44 is this a Sketcher issue or an issue in Curves WB?

AjinkyaDahale, 2025-01-24:
I think this is a curves wb issue. The control points were throwing off some limits. If one can reproduce this without installing the curves wb, that may be a different thing.

maxwxyz, 2025-01-24:
@theosib can you reproduce without the Curves WB? Otherwise please open an issue in https://github.com/tomate44/CurvesWB

closed 2025-01-24 by maxwxyz

tomate44, 2025-01-24:
> I think this is a curves wb issue. The control points were throwing off some limits, so they are removed. If one can reproduce this without installing the curves wb, that may be a different thing.

I confirm. Sketch_On_Surface removes the BSpline control points. This is the intended behaviour.
@theosib , the control points can be restored with Menu Sketch -> Sketcher Visual -> Show/Hide internal geometry.

Alternatively, it is recommended to keep the Mapped Sketch with the blue rectangle alone, and to map additional objects with the Sketch_On_Surface ExtraObjects property.