FreeCAD/FreeCAD #19217: New transform tool and CAM setting origin don't work together

leho-l, 2025-01-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When using the new transform tool to set up  CAM job for the flip side of a double side machined part
setting the origin via the Job setup dialog results in parts not moving or moving not as expected.
Notes: 
The transform tool now doesn't work with multiple bodies selected. They need to be in a part container.
Old workflow was to edit the placement setting with all bodies selected. Now you can't edit placement and transform with more than one selection.
You cant select individual bodies that are in a part container when making a new job.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39971 (Git) Snap 1278
Build type: Release
Branch: main
Hash: 076899205c7b6c3d3ae31e27a3144568d704f132
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: Estonian/Estonia (et_EE)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * PieMenu 1.11.2 (Disabled)
  * Curves 0.6.52
  * Gridfinity.backup1735300935.9070914 (Disabled)
  * sheetmetal 0.6.13
  * btl 0.9.9
  * Gridfinity 0.6.0
  * Freecad-Built-in-themes-beta 1.2.2
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

CAM

### Anything else?

https://github.com/user-attachments/assets/7d8a4bb2-0c47-43b3-a4bc-0479e19973f9

[transform tool in CAM.zip](https://github.com/user-attachments/files/18534184/transform.tool.in.CAM.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-24:
please also share a quick screencast how it looks in 1.0 with the "old" transform tool.

leho-l, 2025-01-27:
In freecad 1.0 it works.
https://github.com/user-attachments/assets/c07a64b2-e4fb-4035-a16b-ff60318bdc63

maxwxyz, 2025-01-27:
In 1.0 you are using the placement command, not the transform tool.

leho-l, 2025-01-28:
Yes, this is not available in this usecase in the 1.1 dev version.
Playing around eit th e dev version 

https://github.com/user-attachments/assets/81082ac4-6b9d-4285-b0c6-cfb0d110c7ab

luzpaz, 2025-01-28:
Hey @kadet1090 can you take a look ? TIA

luzpaz, 2025-05-10:
soft bump

luzpaz, 2025-05-31:
Is this considered a regression ?

tarman3, 2025-09-24:
> Is this considered a regression ?

Yes
New transform tool completely do not work in CAM workbench

tarman3, 2025-09-28:
One note
New transform dialog working fine with operation `PathShape`

tarman3, 2026-02-28:
@leho-l 
Please check last weekly build

luzpaz, 2026-03-26:
@tarman3 do we know what commit might of solved this ticket ?