FreeCAD/FreeCAD #19222: Std_Part unusably small

theosib, 2025-01-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I have historically used the Origin in Std_Part to orient myself with respect to my design. I'll unhide it so that I can see the axes and planes and then I can keep track of things like what plane I want to mirror something by, etc.

In the latest snapshots, the Std_Part Origin is practically microscopic. I can see the axis labels, but I can't tell which is which, because the letters are too small. And there don't appear to be labels on any of the planes.

I've gone hunting through the preferences, and I can see no means to change the sizes of any parts of it, which makes it entirely useless, and this is making it way harder for me to work on some of my designs.

Also, the size of the Origin doesn't change with zoom. And it used to be that if I put an object into the Std_Part, it would scale the planes and axes with the contents of the Std_Part. But this doesn't happen anymore. So there's no means right now to mitigate the problems I'm having with it.

[image]

### Full version info

```shell
OS: macOS 15.1.1
Architecture: arm64
Version: 1.1.0dev.39896 (Git) Conda
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US) [ OS: C/Default (C) ]
Stylesheet/Theme/QtStyle: unset/Classic/Qt default
Installed mods: 
  * freecad.gears 1.3.0
  * Pyramids-and-Polyhedrons
  * fasteners 0.5.29
  * Curves 0.6.50
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

theosib, 2025-01-24:
BTW, the same problem happens in Part Design. The planes have no labels, and the axis letters are unreadably small on the screen. It's readable in the screenshot just because it's zoomed in.

To give you a sense of scale, I'm pasting a shot of the screen.

[image]

theosib, 2025-01-24:
JohnOCFII suggested I add a clarification. The first image here is what you get when creating a sketch on a body in Part Design. This is nice.

[image]

However, if you unhide the origin after creating the sketch, then it's too small.

[image]

obelisk79, 2025-01-24:
@kadet1090 could you look into this?

benj5378, 2025-01-24:
I totally agree with it being too small. When mirroring objects, the shown planes should be sized accordingly to the model as previosuly.

@PaddleStroke

maxwxyz, 2025-01-24:
IMO it needs a setting in the preferences to change the global size for the datum objects. There are scenarios where too big datums are cluttering the view where multiple objects are visible.

benj5378, 2025-01-24:
> There are scenarios where too big datums are cluttering the view where multiple objects are visible.

I am not sure if this scenario actually exists. I have never had more than 3 datums showing at once. I found the previous system with autosizing according to objects very satisfying.

kadet1090, 2025-01-24:
> [@kadet1090](https://github.com/kadet1090) could you look into this?

I'm playing with it already. My initial thoughts are that the default size for planes should be somewhere inbetween the current state and the one used for selection - roughly in the middle. I'm also working on better feedback for plane pre-selection:

https://github.com/user-attachments/assets/ea485d06-c85f-46dd-bbc7-655268369146

This however to work properly requires https://github.com/coin3d/coin/issues/546 to be solved and proper version of Coin to be propagated to FreeCAD (#18883 would solve that issue)

maxwxyz, 2025-01-25:
> I am not sure if this scenario actually exists. I have never had more than 3 datums showing at once. I found the previous system with autosizing according to objects very satisfying.

That's one reason why it was changed. The big datums were always cluttering the view, when trying to use them you had to constantly zoom in and out as all had different sizes. Some of the discussion was happening in https://github.com/FreeCAD/FreeCAD/issues/12584 but there are other issues too. 
The idea is to make the origin system with base plane not as obtrusive when viewing it and to visualize part placement and orientation, especially in assemblies (with multiple parts):

[image]

[image]

Before that change it looked like this with this simple document with 2 objects:

[image]

The new datum objects also have a constant view size, same as the transform dragger already had. Ideally, we have a global preference to adjust the datum scale of such elements, what would change the default size depending on the visible screen space - but not for each tool or object but this should scale all these elements, like datum objects, transform dragger, axis cross,...
I agree, it needs better selectability when the task asks for an input, so datum planes and axis should be larger to better select them.

theosib, 2025-01-25:
I don't care how big the datum planes are. Even the current microscopic ones are large enough to be selected. 

My whole problem is that I can't SEE the LABELS. And I rely on the labels.

kadet1090, 2025-01-25:
> My whole problem is that I can't SEE the LABELS. And I rely on the labels.

Can you describe when you rely on labels, in what scenarios and why they are important in such context?

For selection labels are displayed, datum labels were never labeled from what I remember.

theosib, 2025-01-25:
One example is when I want to mirror along a coordinate plane. I don't necessarily remember the name of the plane. So I show the planes, take mental note of which one is the basis I want for mirroring, unhide the planes, and use the mirror tool. In such scenarios, the bigger the better since I just want to get some information and then hide it again.

Another scenario is when I want to revolve around an axis.

I usually start my designs on the XY plane, so I always remember that, but I always have to check for X and Y (or XZ or YZ).

So let me pose you a question. What is the point in being able to show the planes in the first place if they don't convey any information?

kadet1090, 2025-01-25:
> So let me pose you a question. What is the point in being able to show the planes in the first place if they don't convey any information?

I'd argue that in most cases the visual placement of plane in relation to objects in 3D scene is more useful than remembering planes by the name.

That said, point of my question was to understand the use case, not to undermine it. UX process revolves around understanding how users work with the program, what are the needs and in what scenarios information is useful. Having this information allows us to better understand different workflows, as every user uses a program in a different way. 

Would showing the label on hover (mouse over) work for you? I am afraid that always showing labels for each plane can be a bit overwhelming, but I agree that user should be able to confirm that this is indeed the correct plane by name.

theosib, 2025-01-25:
Any means by which I can quickly check which planes are which is fine by me.

Heck, one option would be to modify that cube in the upper right to say not "Top" but "Top\nXY", and so on.

This would eliminate one of the two main use cases I have for Std_Part. (The other being to group objects into bodies that I can move around with respect to each other.)

JDMils, 2025-02-16:
Here's my comment on this topic. I posted my question here in the forums: [https://forum.freecad.org/viewtopic.php?p=811483#p811483](url)
* I really liked the way 0.21 shows the 3 planes both at the beginning and during the Sketcher process. It gives me the ability to better judge the location and firection of the axis'.
* I personally find the new smaller axis' too difficult to work with. If I have a model centered on the origin, the axis' are now in the middle of the model and if the model is solid then it makes it more difficult to work with the axis'.
* I would like to ask for an option to be provided in Edit | Preferences:
* * Select the new smaller axis.
* * Select the size of the axis. This enables the new axis to be viewed, however gives the user the ability to make them larger just like they were in 0.21. This should be a percentage of the model size. So if the model grows in size then the size of the axis should as well. So, if I set the axis size to 120% of the model size, no matter how big the model gets, the axis will always show 20% larger than the model.

Please bring the axis size back!

OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/Australia (en_AU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * A2plus 0.4.68
  * Assembly4 0.50.15
  * Curves 0.6.54
  * fasteners 0.5.36
  * Gridfinity 0.8.0

Brd-Prey, 2025-02-23:
I just spent hours trying first to make a datum larger and then the axis lines/planes.   Seems it was under properties of a datum if you checked manual size and now its gone.   Please bring back the choice.

maxwxyz, 2025-02-23:
I'm closing this in favor of a size preference

closed 2025-02-23 by maxwxyz

theosib, 2025-03-05:
> I'm closing this in favor of a size preference

I would like to suggest that the font size scale with the size preference, because the X, Y, and Z are too small for me to identify.