FreeCAD/FreeCAD #19224: TechDraw: Export actions duplicated

PaddleStroke, 2025-01-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

If you right click, the context menu shows 3 entries : 
Export DXF
Export SVG
Export PDF

And the toolbar has 2 commands : 
Export DXF
Export SVG

Both of those are implemented separately. The commands are in `Command.cpp` : `CmdTechDrawExportPageDXF`
The context menu actions are defined in `MDIViewPage` : `saveDXF`

So the code is duplicated.

Also the toolbar don't have export PDF. Which admitedly can already be done by "Print all".
Also the commands are opening and commiting a command. Which makes no sense for an export action.

@WandererFan FYI

### Full version info

```shell
1.1
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-24:
Print all is still different I guess, ideally is Export -> Technical Drawing and then change the filetype to .svg or .pdf

maxwxyz, 2026-11-07:
Part of this is fixed in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/506e4e4f846ab42b88ae0244cf565ceea2c84450

The DXF export paths no longer open and commit an editing transaction. The duplicate export implementations are not consolidated by this commit.