FreeCAD/FreeCAD #19228: Please consider visual impairments when making UI design choices

theosib, 2025-01-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I was inspired by https://github.com/freecad/freecad/issues/19222 to make this general plea to FreeCAD developers regarding the sizes of UI elements.

I'm 51 years old and as is normal with someone my age, I have presbyopia. This means I have trouble seeing things that are really small or up close and require reading glasses to do things like read ingredients labels on food packages, etc. One of the first things I did when switching to the FreeCAD weekly builds was increase the font size in the Model view from 11pt back to the 13pt it was in previous versions. I also have up-sized quite a lot of other UI elements to make them easier to see, some because they're too small, some because I don't think FreeCAD does a good job of scaling some things properly to the Apple retina displays (making them too small for ALL users).

Nowadays, people live to 100 years old, and you will have users even older than me who need even stronger reading glasses. Not to mention the millions of people **born** with visual impairments. I'm making this note to advocate for the potentially large portion of your user base who will struggle quite a lot when dealing with UI elements that are too small and will also struggle to figure out where in the settings they have to go to fix these things.

I ask that FreeCAD devs please take eyesight challenges into consideration when making UI design choices. Thanks.

### Full version info

```shell
OS: macOS 15.2
Architecture: arm64
Version: 1.1.0dev.39896 (Git) Conda
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US) [ OS: C/Default (C) ]
Stylesheet/Theme/QtStyle: unset/Classic/Qt default
Installed mods: 
  * freecad.gears 1.3.0
  * Pyramids-and-Polyhedrons
  * fasteners 0.5.29
  * Curves 0.6.50
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

JohnOCFII, 2025-01-24:
One related item is that default user interface elements in FreeCAD often look much smaller on macOS than they do in Linux, even on the exact same hardware.  I'm not sure if this is related to Apple's "Retina" resolution vs. whatever is the default on Linux.

theosib, 2025-01-24:
On the Mac, I like to have the screen real-estate but with key UI elements (like text and those little dots that you use to select corners, etc.) sized up as appropriate. Also, it would be nice if things scaled automatically on the Mac to adapt to the high DPI rather than requiring the user to zoom or edit configs.

obelisk79, 2025-01-24:
> [@obelisk79](https://github.com/obelisk79) is there any place for mentioning this in your guidelines?

I will, but I believe this is more of a technical issue than it is a design issue. FreeCAD, Coin3D, and QtWidgets were originated well before Hi-DPI screens were prevalent and therefore don't handle scaling particularly well. I believe this will improve if we migrate over to the newer QtQuick implementation, but this will be a protracted process that will probably span at least a few major version releases. In the meantime, it's probably best to report individual inconsistencies as they are encountered so they can each be directly addressed.

maxwxyz, 2025-01-24:
This needs to be considered, fully agreed. There are a lot of HiDPI issues already, I cannot think of a single PR fixing all of them at once, so I am closing this single issue.

closed 2025-01-24 by maxwxyz

theosib, 2025-01-25:
I think people agree with me that making sure UI elements are readable is important. I fear that if this issue is just closed then this will get forgotten.

hyarion, 2025-01-25:
> I think people agree with me that making sure UI elements are readable is important. I fear that if this issue is just closed then this will get forgotten.

While important, I don't think an issue is the right place for this.
Issues gets high visibility when new but then it'll be buried after a day or two and no one will see it.

Issues should also be possible to close, otherwise we'll get a never ending pile of them. 

Obelisk is creating guidelines for ui and by including this concern it will be seen every time that is read. I think it will be more visible that way.

If there are specific problems, feel free to create issues for those :)