FreeCAD/FreeCAD #19231: Make PartDesign Hole size parameterizable

alpdigitech, 2025-01-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

PartDesign Hole -> Thread Size should be able to accept "M14" or "14 mm" and can be an expression that could reference variables from Var Set or Spreadsheet. Hole -> Threaded and Hole -> Model Thread should also be parameterizable such as for the use case of turning them to True only at the final stages to speed up development.

Right now everything in my model resizes properly except for those Metric Holes. I don't want to tinker with macros.

### Full version info

```shell
FreeCad 1.0.0
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-25:
@alfrix FYI

alfrix, 2025-01-25:
Any checkbox is can be assigned with an expression, it's just not doable from the taskpanel, you need to do it from the properties, and you would assign it to a bool in the varset

It can be added to taskpanels with a custom widget with an Fx logo or maybe a context menu on right-click on the checkbox, 
He is requesting to be possible, and it already is

The other is an enumeration, you cannot enter direct values on enum, the procedure would be the same just that i think varset doesn't support it at the moment #15553

but both of this are more of a general Ui thing not a FeatureHole specific thing.

alpdigitech, 2025-01-26:
Thank you @alfrix for the pointer. This is what I did: create a variable in var set named "screw_id" then assign it the expression "<<M%d>> % (screw_radius*2/1mm)". In the PartDesign Hole part, right click on the "Hole -> Thread Size" property and click "Expression..." and then assign it "<<var>>.screw_id". It works!

tables00, 2026-09-02:
> Thank you [@alfrix](https://github.com/alfrix) for the pointer. This is what I did: create a variable in var set named "screw_id" then assign it the expression `<<M%d>> % (screw_radius*2/1mm)`. In [...]

Tried this workaround in 1.1.3, and it seems like it doesen't work, as it just ends up with M1x0.25. Did you do anything else aswell?

[image]

[image]

Chiming in here just in case someone has an easy fix for this while waiting for the feature.