FreeCAD/FreeCAD #19235: Sketcher: Won't let user pretend to make a redundant constraint

theosib, 2025-01-25:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In the past, I have regularly measured things in Sketcher by taking steps to create a redundant constraint just to see the value, and then I cancel.

But now Sketcher just gives me an error when I try to do that, so I can't double check constraints.

Example:
See the screenshot. What I want to do is find out the vertical distance between the center of the square and the imported corner in the upper left.
But since this would be a redundant constraint (if I actually decided to apply it, which I won't), then it won't even tell me what the constraint would be.

I notice that if I go to apply a constraint to external geometry, it shows me what that constraint would be, just in blue. It sure would be nice if I could do the same thing for redundant constraints between internal elements.

[image]

### Full version info

```shell
OS: macOS 15.2
Architecture: arm64
Version: 1.1.0dev.39896 (Git) Conda
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US) [ OS: C/Default (C) ]
Stylesheet/Theme/QtStyle: unset/Classic/Qt default
Installed mods: 
  * freecad.gears 1.3.0
  * Pyramids-and-Polyhedrons
  * fasteners 0.5.29
  * Curves 0.6.50
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

theosib, 2025-01-25:
The bottom line is that it's useful to have some means in Sketcher to measure between things that are already constrained. Creating a fake constraint (like what happens with imported geometry) would be just as good as any other means.

Roy-043, 2025-01-25:
You can just create a reference dimension.
https://wiki.freecad.org/Sketcher_ToggleDrivingConstraint

IMO this can be closed.

pierreporte, 2025-01-25:
Sometimes making redundant constrains can be useful, when altering a sketch to change how it is constrained. I do have the same problem, despite the “auto remove redundants” option toggled off.

casterle, 2025-01-25:
In FreeCAD 1.0 if you try to create a conflicting constraint (other than on external geometry), the distance dialog box opens so that you can create a reference constraint. If you select external geometry and then try to create a constraint FC will automatically create a reference constraint.

If you try to create a conflicting constraint in the 1.1.39980 dev version (other than on external geometry), you are not offered a dialog box so you can create it as a reference constraint. Rather, you get an error "Not allowed to edit the datum because the sketch contains conflicting constraints.". 

After dismissing the warning, a dialog box does not appear and the sketcher remains in the data mode. Selecting another object or right-clicking to exit data mode causes the constraint you are trying to create to vanish.

I would suggest that the behavior from 1.0 (where you have an option to create a reference) be restored. Alternatively, and more consistently, a reference constraint could be automatically created regardless of the type of geometry selected.