FreeCAD/FreeCAD #19238: Compatibility issue 0.21.2 vs 1.0.0: Chamfer "Two distances" option

szdowk, 2025-01-25:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

There is a problem with chamfer while opening 0.21.2 files in Freecad 1.0.0.

If the option "Two distances" was used in 0.21.2 while creating a chamfer, then parameters "Size" and "Size2" are swapped during recomputation when the file is opened in 1.0.0.
Values of parameters are imported appropriately as "Size" and "Size2" but used vice versa.

To reproduce:
1. In 0.21.2 start Part Design module.
2. Draw a box in sketcher.
3. Pad it.
4. Select any face and use chamfer function.
5. In chamfer dialog select "two distances" option and input two various values as sizes.
6. Save
7.  Open the saved file in 1.0.0, recompute object.

[image]

[image]

### Full version info

```shell
0.21.2 rev. 33771
1.0.0 rev. 39109
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

szdowk, 2025-01-25:
OS: Slackware 15.0 x86_64 (KDE/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * sheetmetal 0.6.13
  * Help 1.0.3
  * freecad.gears 1.3.0
  * Curves 0.6.53

OS: Slackware 15.0 x86_64 (KDE)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * sheetmetal 0.6.13
  * Help 1.0.3
  * freecad.gears 1.3.0
  * Curves 0.6.53

szdowk, 2025-02-09:
Similar problem appears when there is used option "Distance and Angle" to create chamfer.

[image]

[image]

szdowk, 2025-02-12:
I compared chamfers in PartDesign and Part modules.

For Part module, the behavior of direction of chamfers in 1.x and 0.21.2 is the same and correspond to behavior of PartDesign module 1.x.

I'm just curious if PartDesign chamfer was broken all the time in 0.21.2 and older versions ?!

Anyway, there is incompatibility, that brakes models from Freecad 0.21.2 in Freecad 1.x.

If the Freecad 1.x works properly, it should flip direction of PartDesign chamfers while opening older files.

TH1813254617, 2025-12-11:
IIRC, this is far from the only incompatibility between 0.21.2 and 1.0.

For me, I had to fix a lot of my files or outright remake them. I remember 1.0 breaking a LOT of my sketch attachments for whatever reason.