FreeCAD/FreeCAD #19239: Spreadsheet wird in den Zellen schwarz nicht angezeigt

Chris1807, 2025-01-25:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Schwarzer Text wird in den Zellen nur in Punkten und fast Unsichtbar angezeigt.
Hintergrund ist weiß eingestellt.

### Full version info

```shell
Version 1.0.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-01-25:
Please translate title and body of this ticket. Please also add your Full [About](https://wiki.freecad.org/About) info

Chris1807, 2025-01-25:
OS: macOS 15.0
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

luzpaz, 2025-01-25:
Again, please translate the ticket and title in to english, please

closed 2025-01-25 by Chris1807

luzpaz, 2025-01-25:
https://github.com/FreeCAD/FreeCAD/issues?q=state%3Aopen%20label%3A%22Mod%3A%20Spreadsheet%22%20color