FreeCAD/FreeCAD #19246: Assembly Belt Joint causes a crash

FredLaCasemate, 2025-01-25:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

On Linux, the belt joint crashes FreeCAD if the two axes are concentric. I have not found another way to assemble two rotating concentric body with a ratio in their rotation speed.
The bug exists as well in the stable 1.0.0 version and in the 39896 built.
In the joined sample I have 3 bodies (2 pulleys and 1 axe). If I try to add a belt joint between the 2 pulleys, Freecad crashes.

[BugBelt.FCStd.tar.gz](https://github.com/user-attachments/files/18539531/BugBelt.FCStd.tar.gz)

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39896 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: https://github.com/FreeCAD/FreeCAD/commit/a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-26:
@PaddleStroke FYI

PaddleStroke, 2025-11-25:
@aiksiongkoh there is also this special 'coaxial gear' joint that needs attention. Especially since it's crashing.

luzpaz, 2026-03-13:
bump @aiksiongkoh

closed 2026-08-27 by maxwxyz