FreeCAD/FreeCAD #19247: Cannot import dxf file from LibreDWG dwg2dxf

ywiyogo, 2025-01-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I got a CAD file in dwg format. Then, I converted it to dxf using the dwg2dxf binary of LibreDWG. After I got the dxf file and open it in FreeCad 1.0.0, I get this error:
```
13:40:46  Found type 999 record when expecting start of a SECTION or EOF
13:40:46  Unable to parse value '
', using zero as its value
13:40:46  CDxfRead::get_next_record() Failed to get integer record type from '
'
```

I don't know if the root cause is from LibreDWG or FreeCAD.

### Full version info

```shell
OS: CachyOS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-01-26:
We need to see the file, can you zip it and upload it to github? (drag/drop into a textwindow of this ticket)

ywiyogo, 2025-01-26:
Do you want to see the both files or only the converted dxf?

luzpaz, 2025-01-26:
Might as well add both.

ywiyogo, 2025-01-27:
[DesignRumah.zip](https://github.com/user-attachments/files/18552337/DesignRumah.zip)
Please check the files.

Pedro-Leitek, 2025-01-29:
Hi there,

I'm facing the same issue for the libredwg. Also the oda converter is throwing errors.

Regards

yorikvanhavre, 2025-02-05:
This was normally fixed in #16131 ... @kpemartin could you have a look here?

kpemartin, 2025-02-06:
The issue with the 999 records should indeed have been fixed by #16131 but I suppose there may be some context where weeding out this record type has been missed, either by accident or by that fix being too strict about where these comment records might occur. My DXF reference says this record type can be used "to include comments in a DXF file" without saying where they can or cannot be used in the file.

I'll have a look at it.

kpemartin, 2025-02-06:
When I try to reproduce the problem I do not get the "Found type 999 record when expecting start of a SECTION or EOF" message, but I get the other two except they show empty quotes rather than a quoted line break. This is likely because the file contains CR/LF line endings, I'm on Windows so these are not returned by `std::getline`, but perhaps on non-windows `std::getline` might leave the CR on the end of the returned string and it is showing up in the error message.

Other than the mystery of why I'm not getting the "999" message, the problem is that the DXF file is defective, and dwg2dxf is at fault.
For example, starting at line 207 in the DXF file:
```
  9
$DIMSHO
 70
     1
  9
$DIMPOST

  9
$DIMAPOST

  9
$DIMALT
 70
     0
```
Those single blank lines after `$DIMPOST` and `$DIMAPOST` are the source of the problem. There should be a line pair there, like the 70/1 after `$DIMSHO` but perhaps with some code other than 70 depending on the data type. These particular variables are described as being General and Alternate dimensioning suffix and should use code 1 (text) for their values

Further down in the file `$DIMBLK1` and `$DIMBLK2` (first and second arrow block names), `$DIMLDRBLK` (arrow block name for leaders), and about a dozen more are similarly broken.

This does not only occur in the HEADER (variables) section, but I've also found a LTYPE record whose "descriptive text", which should be a type 3 record, is missing its record type, and in the STYLE section, at which point I gave up any hope of manually correcting the file to see if there were any other issues.

Just a complete wild guess is that dwg2dxf has started using null pointers rather than empty strings, and the code that puts these values out to the file doesn't know what type code to put out because there is no object to query the type of. So the code just skips the type code and the null string reference goes out as an empty line, Though the fact that some of these should use code 1 and some code 3 would imply that it is impossible to infer the record type from the type of the value.

kpemartin, 2025-02-06:
Just to see if there were other problem with the file I temporarily changed the reader to barge past the errors but gave up after over an hour when there were already over 150,000 objects, and since the code I'm working with does not include #18676 it was likely to take a lot longer to finish.

yorikvanhavre, 2025-02-06:
Anyone willing to upstream this issue to LibreDWG at https://savannah.gnu.org/bugs/?group=libredwg ? LibreDWG seems actively maintained ( https://git.savannah.gnu.org/cgit/libredwg.git ) so there are good chances someone will look at it.

FEA-eng, 2025-11-23:
[This](https://github.com/FreeCAD/FreeCAD/issues/19247#issuecomment-3324899831) is still relevant

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.2.0dev.20251119 (Git shallow)
Build date: 2025/11/19 23:28:17
Build type: Release
Branch: main
Hash: deab0abf8e61d69f2208b794f64ac56428467fef
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```