FreeCAD/FreeCAD #19248: Box selection appearence in skecher

mgth, 2025-01-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Box selection in sketcher has different behavior if you drag left to right (select contained objects) or right to left (select intersecting objects) I thought the apparence would reflect this, as it may be drawn with or without a background. But this gets inverted if the cursor is moving up to down or down to up without changing selection mode. 

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39980 (Git) Conda
Build type: Release
Branch: main
Hash: 836ad91f3ddd3ec63b84b8e3f87a767d5465261d
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: ProDark.qss/ProDark/Fusion
Installed mods: 
  * Assembly3 0.12.2
  * CurvesWB 0.6.12
  * Defeaturing 1.2.3
  * ExtremeProDark 2.6.4
  * FreeCAD-themes 2024.7.24
  * freecad_streamdeck_addon 0.1.7
  * OpenDark 2024.3.13
  * OpenTheme 2024.9.1
  * parts_library
  * parts_library.bak
  * ProDarkThemePreferencePack 1.0.0
  * workfeature
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-27:
@FreeCAD/design-working-group any idea how this should be displayed? Different cursor icon, background of the selection rectangle,...?

PaddleStroke, 2025-08-21:
I didn't even knew about that feature.

closed 2025-08-31 by kadet1090