FreeCAD/FreeCAD #19250: "Duplicate selected object" puts new object in wrong Std_Part

karamellpelle, 2025-01-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Attached is a minimal example of the problem. I start with a _Std_Part_  `Part` containing a (empty) Body with an (empty) Sketch. This _Std_Part_ is duplicated into `Part001` using _Edit > Duplicate selected object_, so we have now `Part` and `Part001`.

If I now make `Part001` the active object in the Tree view, and perform _Edit  > Duplicate selected object_ on its Sketch, the new Sketch is placed in the old `Part`, not the current.

[DuplicateIntoWrongStdPart.zip](https://github.com/user-attachments/files/18549865/DuplicateIntoWrongStdPart.zip)

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39896 (Git) Conda
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * Assembly4 0.50.15
  * fasteners 0.5.33
  * freecad.gears 1.3.0
  * IconThemes
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

karamellpelle, 2025-01-26:
The visualization of active object in _Tree view_ is not very clear, the font is just a bit bolder with an underscore. I've seen more people finding this visualization difficult to see.

And you can obviously both have an active _Part_ and an active _Body_ in the _Tree view_ that are not related.

maxwxyz, 2025-01-26:
@FreeCAD/design-working-group what is the best workflow for this use case?

luzpaz, 2025-09-29:
@karamellpelle can you confirm ?

karamellpelle, 2025-11-12:
(I came back to this issue since I experienced the problem again).

Duplicated objects should be placed along the original source, independent of which Body or Part that is active. Today I duplicated a sketch in a Body in order to attach it on a different face. I got error messages because the sketch couldn't use/find external references. This happened since the sketch became placed outside the Body. The natural behavior would be to place it along the source. These errors should therefore have been avoided.

This is bad UX that contributes to a bad FreeCAD experience for beginners.

FlachyJoe, 2026-03-12:
So the duplicate action respect the active body but not the active part. As said above an active body can reside outside the active part so it's unclear what should be done.

[image]

Maybe this could be split in two issues:

- Active part is not respected by duplicate object (true even if there is no body).
- Active body can be out of active part (active part should be switched to the body's parent one or deactivated if body doesn't reside in a part and body should be deactivated if it's parent part is deactivated)