FreeCAD/FreeCAD #19254: PartDesign: Dimensioned Holes Model Thread fails

pmjdebruijn, 2025-01-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Using the hole tool, the thread isn't modeled unless the hole goes through the model entirely.

[1] (GOOD) If Depth is set to 'Through all' we do see a modeled thread.
[2] (GOOD) If Depth is set to 'Dimension' and the dimension exceeds the model, we do see a modeled thread.
[3] (BAD) If Depth is set to 'Dimension' and the dimension does not exceed the model, we do NOT see a modeled thread.

**HOW TO REPRODUCE**
On an object that is 25mm thick, place a hole using the following parameters:

Profile: ISO metric regular profile
Size: M10
Diameter: 8.53mm (greyed out)
Depth: Dimension, 15.00mm
Reversed, Tapered (both unchecked)
Threaded (CHECKED)
Class: 4G
Direction: Right hand
Model Thread, Update thread view (BOTH CHECKED)
Depth: Holed depth (15.00mm greyed out)
Custom Clearance (unchecked)
Hole Cut Type: None
Drill Point: Angled 118.00 (degrees)
Take into account for depth (unchecked)

If fixed, I hope to see this backported to releases/FreeCAD-1-0...

### Full version info

```shell
FreeCAD_weekly-builds-39980-conda-Linux-x86_64-py311.AppImage

1.1.0dev
39980 (Git)
836ad91 (hash)

(also reproducible on releases/FreeCAD-1-0 specifically 1.0.0+51~g7dca20f979)
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-27:
Please add full about info.

@alfrix @FlachyJoe FYI

alfrix, 2025-01-27:
duplicate of #18796