FreeCAD/FreeCAD #19255: "BRep_API: command not done" is not a clear or actionable error message

wez, 2025-01-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

As someone new to this sort of CAD tool in general and FreeCAD in particular, I find it frustrating that seemingly basic tools such as Thickness, Chamfer and so on can produce the inscrutable "command not done" error and don't provide any context on what the underlying issue might be.

From my own searching I can see that one very common cause is the relationship with wall thickness.

It seems to me that it would be valuable to bubble up this context from the underlying operation and provide some directive advice to the user: in other words, if it is due to wall thickness, include that in the message and suggest an appropriate course of action, and/or output a link to an FAQ or wiki page or something that can explain it in more detail.

### Full version info

```shell
OS: Ubuntu Core 22 (KDE/plasmawayland/xcb)
Architecture: x86_64
Version: 1.0.0.38641 +468 (Git) Snap 1248
Build type: Release
Branch: tag: 1.0.0
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * fasteners 0.5.36
  * FreeCAD-themes 2025.1.7
```

### Subproject(s) affected?

None

### Anything else?

I understand that these tools are likely not at all simple in their implementation, so please don't construe this as me being critical of how they work and what they are capable of. 

I also recognize that it may not be possible for a tool to tell you precisely why it didn't work out. If that is the case, I think it would still be worthwhile to suggest at the most likely reason in the error message that results from the failed operation, so that the user experience is a bit more smooth.

Thank you for creating this software: I think that it is amazing despite having a fairly brutal learning curve.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-01-26:
Please provide a step-by-step way to produce the error for testing convenience, TIA

maxwxyz, 2025-01-26:
This error can be produced by many tools, e.g. fillets with too large radius.
Does anyone know if there is an error message for each command or is this a "core" message which is displayed?

FredLaCasemate, 2025-01-31:
Hi,
As soon as there is a sphere it becomes impossible to create a shell. Here is a simple example:

[image]

If we try to create a shell on the bottom face, FreeCAD fails and we get the message "Command not done"

[image]

This command fail as well in Part as in PartDesign.

Regards,
Fred

Syres916, 2025-01-31:
To the best of my knowledge this is being generated by OCCT and just passed through as an exception.

FlyinCow, 2025-05-22:
> [@luzpaz](https://github.com/luzpaz) simple reproducer:
> 
> 1. Open the 'part design' workbench
> 2. Draw a square on a plane
> 3. Pad to 10mm
> 4. Try to add an 11mm fillet on the padded edge
> 
> [image]

Unfortunately, OCC can't make such fillet. As far as I know, only most common simple situations are supported. For example:

[image]

You can't make a chamfer/fillet wider than 6.84mm on the blue edge (it might be hard to see clearly). Imagine you need to do so, OCC has to create a new face under your chamfer face which terribly breaks the topology relations near the edge. Same thing happens on face offset when you offset faces too far away so some faces should be deleted. OCC can't handle such local topological operations.

redfast00, 2025-05-22:
@FlyinCow I understand why the tool can't do it, I guess the issue is more that the error message you get as a user is not really directly understandable or actionable ('BRep_API: command not done') and could be more userfriendly (for example: "Could not create fillet with radius 11 mm due to ...; biggest fillet radius possible is 8 mm")

FlyinCow, 2025-05-23:
@redfast00 Totally agree, but I think it's hard to generate such diagnostic message which includes accurate length. But it should not be a problem to show which API that throws the exception.

FlachyJoe, 2025-11-10:
I've proposed to mention the possible problematic properties in the error messages. The error mechanism is currently too light it should be reworked to allow the devs to join additional information.