FreeCAD/FreeCAD #19257: Setting sketch property "Show plane" to true before creating geometry removes the sketcher X/Y/root point

macdroid53, 2025-01-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When creating a new sketch, if "Show plane" is set true before any sketch geometry exists:

- upon entering the sketch
--the x/y/root coordinate (red/green axis) is not displayed
--in this condition any attempt to create geometry retuns:
`10:09:34  pyException: Traceback (most recent call last):
  File "<string>", line 4, in <module>
<class 'NameError'>: name 'nan' is not defined
10:09:34  Unnamed#Sketch: Failed to add circle
10:09:34  Tool execution aborted`

Closing the sketch, setting "Show plane" false, reentering the sketch, returns to normal axis display and geometry can be created.

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.39984 (Git)
Build type: Release
Branch: main
Hash: ab311f1f9b43b6e0886283a05abf27a4bbbec1b2
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * fasteners 0.5.35
  * OpenTheme 2024.9.1
  * Defeaturing 1.2.3
  * Silk 0.1.5
  * CurvedShapes 1.0.13
  * sheetmetal 0.6.13
  * DynamicData 2.70.0
  * lattice2 1.0.0
  * Curves 0.6.53
  * Manipulator 1.5.7
  * kicadStepUpMod 11.4.4
  * freecad.gears 1.3.0
```

### Subproject(s) affected?

Sketcher

### Anything else?

Reference #19205 #19206 #18910 

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-26:
@kadet1090 FYI

kadet1090, 2025-08-02:
@macdroid53 Is it still relevant? I cannot reproduce it anymore.

hyarion, 2025-08-13:
@macdroid53 friendly ping :)