FreeCAD/FreeCAD #19258: Mouse navigation style for 3D Studio Max

vault-thirteen, 2025-01-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

FreeCAD 1.0 has no mouse navigation style ( https://wiki.freecad.org/Mouse_navigation ) as in modern versions of 3D Studio Max editor.

In the 3ds Max 2024 following default controls are used.
Panning: MMB.
Orbiting: Alt + MMB.
Zooming: Mouse Scroll Wheel.

### Full version info

```shell
1.0.0.
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-01-26:
Note: for anyone interested in adding this the following files should provide some orientation on implementation:

* https://github.com/FreeCAD/FreeCAD/blob/main/src/Gui/NavigationStyle.h
* https://github.com/FreeCAD/FreeCAD/blob/main/src/Gui/BlenderNavigationStyle.cpp
* https://github.com/FreeCAD/FreeCAD/blob/main/src/Gui/CADNavigationStyle.cpp
* https://github.com/FreeCAD/FreeCAD/blob/main/src/Gui/GestureNavigationStyle.cpp
* https://github.com/FreeCAD/FreeCAD/blob/main/src/Gui/InventorNavigationStyle.cpp
* https://github.com/FreeCAD/FreeCAD/blob/main/src/Gui/MayaGestureNavigationStyle.cpp
* https://github.com/FreeCAD/FreeCAD/blob/main/src/Gui/OpenCascadeNavigationStyle.cpp
* https://github.com/FreeCAD/FreeCAD/blob/main/src/Gui/TinkerCADNavigationStyle.cpp
* https://github.com/FreeCAD/FreeCAD/blob/main/src/Gui/TouchpadNavigationStyle.cpp
* https://github.com/FreeCAD/FreeCAD/blob/main/src/Gui/NavigationStylePyImp.cpp

luzpaz, 2025-01-26:
@vault-thirteen can you contribute or know anyone that is will to ?

vault-thirteen, 2025-01-26:
@luzpaz I do not write in C++ at the moment. My knowledge of C++ is stuck in the Standard C++ as of 1998. 😂

luzpaz, 2025-01-28:
@Rexbas any interest in implementing this ?

Rexbas, 2025-01-28:
No, I don't think we need more navigation styles in the main repo. Instead I think we should do some refactoring so a navigation style can be described using a json file or something like that. That way we can get rid of the 10 times mostly duplicate code we have now and also make navigation more customizable. Besides, if a navigation style can be described using a file then maybe they can be addons.

I have given a navigation refactor some thoughts but I don't have much time to do a massive navigation refactoring now. I can make a new issue to link all navigation related issues and feature request that may be resolved after a refactor to keep some overview and where some refactor ideas can be shared.

luzpaz, 2025-01-28:
>  I can make a new issue to link all navigation related issues and feature request that may be resolved after a refactor to keep some overview and where some refactor ideas can be shared.

Yes, that would be fantastic.