FreeCAD/FreeCAD #19260: No way to customise accent colour of pressed button

vault-thirteen, 2025-01-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In FreeCAD 1.0 there is no way to customise the accent colour of pressed buttons.

The empty colour theme uses light blue colour as accent colour for pressed buttons, the built-in Light and Dark themes use some greyish colours which are hard-coded into the editor, and they are actually not really distinguishable at all. 

It is not convenient to work with hard-coded colours when many parts of the editor already have customisable colours. Also, it is possible to change accent colour of pressed buttons in the 3ds Max editor. 

Thank you.

### Full version info

```shell
1.0.0.
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-27:
Please add full about info. Which theme do you use?
IMO it is not necessary to expose each and every color as preference to the user @FreeCAD/design-working-group

MisterMakerNL, 2025-01-27:
The stylesheets are copy of Qtstyle fusion on windows that doesn't use color on buttons, I think the windows Qtstyle does use a blue color tint. Personally I don't like extra color, the border color that there is to indicate the state of a button is for me sufficient.
I do remember when I was checking that glow out, that I got issues with color banding. The blue is very subtle and uses some kind of gradient, which is very hard to replicate.

If more people prefere to glow effect instead of the border effect I can try and make it work.

vault-thirteen, 2025-01-27:
> Please add full about info. Which theme do you use? IMO it is not necessary to expose each and every color as preference to the user @FreeCAD/design-working-group

I am not using a single theme. My question is about all the themes at the same time. I tried to select a theme and customise the button colour, but I found out that it was not possible.

Here are the screenshots of some panel in different themes.
I set theme this way: 
Menu `Edit` -> `Preferences ...` -> then `Display` section -> `UI` -> `Theme customization`-> `Style sheet (advanced)`

1. `No style sheet` option.

[image]

2. `FreeCAD Light` option.

[image]

3. `FreeCAD Dark` option.

[image]

And below is a screenshot how it looks in 3ds Max 2024.

[image]

I can customise an accent colour of pressed buttons with my colours in 3D Studio Max, it is yellow in this example.

vault-thirteen, 2025-01-27:
> Personally I don't like extra color, the border color that there is to indicate the state of a button is for me sufficient. I do remember when I was checking that glow out, that I got issues with color banding. The blue is very subtle and uses some kind of gradient, which is very hard to replicate.

I understand what you mean. The thing is that not all monitors are the same. On some monitors the differenece between a pressed button and a non-pressed button is barely visible. This is especially important for the `Light` and `Dark` themes. The solution for me is to use high contrast colouring scheme such as the yellow colour in a 3ds Max example above.

Syres916, 2025-01-27:
@vault-thirteen then just search for:

```
QToolButton:checked {
  background-color: qlineargradient(x1:0, y1:0, x2:0, y2:1, stop:0
```

in the qss file you are using and change (purely as an example) to something like:

```
QToolButton:checked {
  background-color: qlineargradient(x1:0, y1:0, x2:0, y2:1, stop:0 #ffff00, stop:0.7 #9d9d00, stop:1 #9d9d00);
```

[image]

MisterMakerNL, 2025-01-27:
Owh you mean those buttons! Well we had a big long discussion about it last year. The color gives issues with our colorful icons so I think therefore it was decided to do a minimal but just enough indentation. 
You can still as @Syres916 do this yourself off-course. So if you want to make it look more like 3dsmax we have to get less colorful icons, and the only way to do this is to make a light and dark icons set. Like 3Dsmax has a lot less colour and uses black a lot, if you do that in Freecad it will not work in the dark theme.

markkomartin, 2025-08-28:
> Owh you mean those buttons! Well we had a big long discussion about it last year. The color gives issues with our colorful icons so I think therefore it was decided to do a minimal but just enough indentation. 

Hi @MisterMakerNL .
Unfortunately the minimal indentation is not visible on the dark theme.

Do you think the following scenario could be possible?
- gray snap icon = snap disabled
- colored snap icon = snap enabled.

I attach an example image with 3 active snaps: endpoint, perpendicular, near.

[image]

markkomartin, 2025-12-29:
Thanks @kadet1090 !
Your change makes the situation much better, and I think this ticket can be closed. 
Do you agree @leoheck ?

kadet1090, 2026-01-01:
Please reapply the theme - it should work properly.

kadet1090, 2026-01-02:
https://github.com/FreeCAD/FreeCAD/pull/26592 should make sure that the new theme is applied automatically - I'm closing this issue as it is solved for some time

closed 2026-01-02 by kadet1090