FreeCAD/FreeCAD #19262: B-spline pockets can break geometry (non-manifold edges)

Scafir, 2025-01-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

It seems like b-splines pockets may create a bunch of errors in the produced geometry.

I am creating an object by repeatedly pocketting sketches containing a b-spline. It is unclear to me why, but this produces non-manifold edges. Sometimes the issue is visible (extra face added in the model, or produce face is not correct), sometimes not. Because it is sometimes visible, I think there's a problem with the pocket rather than stl/step exporters.

I was unable to create a trivial case showcasing the issue, but I managed to create a simplified model that was still encountering the same problem:  [Wallet_issue_final.zip](https://github.com/user-attachments/files/18550920/Wallet_issue_final.zip)

Checking the geometry within freecad does not find any issues, but importing it in a slicer causes many non-manifold edges errors there:

[image]

### Full version info

```shell
OS: Fedora Linux 41 (Workstation Edition) (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.1.0dev.39980 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 836ad91f3ddd3ec63b84b8e3f87a767d5465261d
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * kicadStepUpMod 11.2.9
  * 3D_Printing_Tools
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-27:
The geometry has no issues:

[image]

It might be depending on your setting for the meshing. Which mesher and settings did you use in the Mesh WB?

Scafir, 2025-01-28:
Thank you for your response. I did not use the mesh WB per se, as I used the File->export menu. 

I am not sure if I trust the result of the checker, as on the [full model](https://github.com/user-attachments/files/18578978/Wallet2_backup_through.zip), I can clearly see that something bad is happening to the displayed geometry (the highlighted face is not supposed to poke through the triangular pocket:

[image]

Creating a mesh from shape with the mesh WB and default settings (0.1mm surface deviation, 30° angular deviation), the geometry is clearly incorrect:

[image]

Same thing with the simplified geometry model:

[image]