FreeCAD/FreeCAD #19263: Toolbars without a name

vault-thirteen, 2025-01-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

FreeCAD 1.0.
I have two toolbars without name.
How do I know what they are ?
Why do they have no name ?

[image]

Thank you.

### Full version info

```shell
1.0.0.
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

tarman3, 2025-01-26:
Try restart FreeCAD

vault-thirteen, 2025-01-26:
The more I use the editor, the more unnamed toolbars I get.
This is what I have after several restarts.

[image]

xtemp09, 2025-01-26:
@vault-thirteen, could you provide _Help_ → _About_ info?

vault-thirteen, 2025-01-26:
> OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

xtemp09, 2025-01-26:
They are the toolbars at the bottom-right corner! They are duplicated and nameless! 😃

vault-thirteen, 2025-01-26:
> They are the toolbars at the bottom-right corner! They are duplicated and nameless! 😃

So, there are four of them already. 🤣
Please, fix this bug.
Thank you.

maxwxyz, 2025-01-26:
I cannot reproduce, could you do Help>Restart in safe mode and check is present?

vault-thirteen, 2025-01-26:
In the safe mode I see only two unnamed toolbars.

[image]

furgo16, 2025-01-26:
Issue for the duplicate toolbars: https://github.com/FreeCAD/FreeCAD/issues/17044

Syres916, 2025-01-27:
Having tested Qt5.15.2 (a few years old) on Windows 7 which produces the exact same issue, I'd say this is a Windows for Qt specific issue.

GAZ082, 2025-01-27:
This happened to me in Linux as well.

Jonezzzzz, 2025-01-27:
This seems to be already fixed with PR #18064.

maxwxyz, 2025-01-28:
So can the issue be closed?

vault-thirteen, 2025-01-28:
> So can the issue be closed?

Sure.

closed 2025-01-28 by vault-thirteen