FreeCAD/FreeCAD #19265: Left handed threads wrong on fine threaded bolts

jonasohrnell, 2025-01-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When using fasteners WB and creating ISO 8676 or ISO 8765 bolts in the dimension M20x1.5, the bolts come out OK when using normal right handed threads, but when threads are changed to left handed the bolt gets messed up - looks inverted, like a pipe with threads on the inside. 

Other bolts seem OK. Some other dimensions of the ISO 8676 and ISO 8765 bolts are OK, but not all of them.

Replication:
- new project
- choose Fasteners WB
- choose ISO 8676 bolt
- choose dimension M20x1.5
- choose threads True, Left handed True

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39980 (Git) Conda
Build type: Release
Branch: main
Hash: 836ad91f3ddd3ec63b84b8e3f87a767d5465261d
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Swedish/Sweden (sv_SE)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * fasteners 0.5.36
```

### Subproject(s) affected?

None

### Anything else?

[FastenersBoltLeftHandedBug.zip](https://github.com/user-attachments/files/18551310/FastenersBoltLeftHandedBug.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

tarman3, 2025-01-26:
You should create issue in fasteners WB page

maxwxyz, 2025-01-26:
Please open an issue here: https://github.com/shaise/FreeCAD_FastenersWB

closed 2025-01-26 by maxwxyz