FreeCAD/FreeCAD #19268: Importing an exported DXF results in an invisible object

vault-thirteen, 2025-01-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I have drawn a simple 2D drawing consisting of four big rectangles and a single small rectangle, representing a basic 2D building floor. Then I exported this drawing to DXF format. I checked this file in an external viewer, DXF file is good. When I try to import this DXF file into the same FreeCAD which exported it, I see only invisible objects. The mouse shows that the object is there, but it is invisible for some reason. Here is the screenshot.

[image]

If I import this DXF file in safe mode, the drawing is not invisible, i.e. I can see it !

[image]

DXF settings are the same in both normal and safe modes.

[image]

Why is it so ? Why is the drawing invisible ?

There is a warning after the importing. It may be related to the issue.

[image]

The DXF file is attached.

[1.dxf.zip](https://github.com/user-attachments/files/18551668/1.dxf.zip)

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

vault-thirteen, 2025-01-26:
I am sorry. It was my mistake. I checked the `Use colors from the DXF file` settings and it created a drwaing with white colours.
But the warning does not disappear. Is this warning important ?

vault-thirteen, 2025-01-26:
I have an idea. The process of importing should compare background colour set in the editor with the DXF settings. If all the drawings in the DXF file are using white colour and if the editor also uses the white colour, a warning should be shown to the user.

maxwxyz, 2025-01-27:
Closing this. Should be an issue of the DXF exporter

closed 2025-01-27 by maxwxyz

vault-thirteen, 2025-01-27:
I wanted to change the issue text but found it closed. Should I open a new issue or try to add new description ?

vault-thirteen, 2025-01-27:
You have not responded in an hour, so I will open a new issue instead of changing the text of this issue.