FreeCAD/FreeCAD #19270: Illegal storage access... - when using make_text() and FreeCADGui.updateGui()

brettersrebug, 2025-01-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I am using python to create a FreeCAD 3D drawing using PyCharm. I create different kind of objects of 'Part' and 'Draft' workbench - everything works fine except Draft.make_text(). I created a font.conf file and added it to the environmental variables - FONTCONFIG_PATH=C:\tools\FreeCAD_100\bin\etc\fonts - to avoid 'Fontconfig error: Cannot load default config file' (see attached files font.conf.txt (renamed to txt for upload)...but also without this font.conf file it terminates because of 'Illegal storage access...'.

For better performance I use the headless mode but for "user-friendly format" I need to create texts. This is not possible in headless mode because ViewObject is None. Therefore I set in my test code 'use_headless_mode = False' and 'use_make_text = True' - and then I get the error when executing FreeCADGui.updateGui():
        #  Illegal storage access...
        #  Terminating...
        #  Abnormal program termination...
try-except does not help here.

The line print(f"Save successful: {file_path}") is never reached.

```
import FreeCAD
import FreeCADGui

use_headless_mode = False
use_make_text = True

if not FreeCAD.GuiUp and not use_headless_mode:
    FreeCADGui.showMainWindow()

import Draft

doc_name = 'DraftTextExample'
doc = FreeCAD.newDocument(doc_name)
file_path = "c:/temp/_aDraftTextExample.FCStd"
doc.saveAs(file_path)

FreeCAD.setActiveDocument(doc_name)

if hasattr(FreeCADGui, 'ActiveDocument'):
    # Update the 3D view
    if FreeCAD.GuiUp:
        FreeCADGui.updateGui()  # is ok

    while not FreeCAD.GuiUp:
        time.sleep(1)

pl = FreeCAD.Placement()
pl.Rotation.Q = (0.0, 0.0, 0.0, 1.0)
pl.Base = FreeCAD.Vector(-44.24086307567845, -13.938928557833574, 0.0)
rec = Draft.make_rectangle(length=27.755539650231462, height=7.8289007267484525, placement=pl, face=True, support=None)
Draft.autogroup(rec)
FreeCAD.ActiveDocument.recompute()
doc.saveAs(file_path)  # is ok
if FreeCAD.GuiUp:
    FreeCADGui.updateGui()  # is ok
doc.saveAs(file_path)  # is ok

if use_make_text:
    pl = FreeCAD.Placement()
    pl.Rotation.Q = (0.0, 0.0, 0.0, 1.0)
    pl.Base = FreeCAD.Vector(-39.73873221634025, -31.476950998341813, 0.0)
    _text_ = Draft.make_text(["test text"], placement=pl)
    if FreeCAD.GuiUp:
        _text_.ViewObject.FontSize = 50
        _text_.ViewObject.FontName = "Segoe UI Black"
        _text_.ViewObject.TextColor = (0.0, 0.0, 1.0)
    Draft.autogroup(_text_)
    FreeCAD.ActiveDocument.recompute()
    if FreeCAD.GuiUp:
        FreeCADGui.updateGui()  # ERROR!!! Not OK - gives:
        #  Illegal storage access...
        #  Terminating...
        #  Abnormal program termination...

doc.saveAs(file_path)

#  Error: saving without updateGui() before gives:
#  (<class 'OSError'>, OSError({'sclassname': 'class Base::AccessViolation', 'sErrMsg': 'Access violation', 'sfile': '', 'iline': 0, 'sfunction': '', 'swhat': 'Access violation', 'btranslatable': False, 'breported': False}), <traceback object at 0x00000215A920D400>)

print(f"Save successful: {file_path}")
FreeCAD.closeDocument("DraftTextExample")
FreeCAD.setActiveDocument("")
FreeCAD.ActiveDocument = None
if FreeCAD.GuiUp:
    FreeCADGui.updateGui()
```

python code:
[freecad_make_text_illegal_storage_access.py.txt](https://github.com/user-attachments/files/18554343/freecad_make_text_illegal_storage_access.py.txt)

fonts.conf:
[fonts.conf.txt](https://github.com/user-attachments/files/18554312/fonts.conf.txt)

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Austria (de_AT)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-27:
@Roy-043 any idea?

brettersrebug, 2025-01-28:
I put the code to a file c:\temp\free_cad_make_text.py and executed via command line (using **a fresh freecad portable version**):

C:\temp>"C:\tools\FreeCAD_100\bin\python.exe" free_cad_make_text.py
Fontconfig error: Cannot load default config file
Illegal storage access...
Terminating...
Abnormal program termination...

then **removed the FreeCADGui.updateGui() after make_text** - now I get the failure on doc.saveAs() - maybe this helps?

C:\temp>"C:\tools\FreeCAD_100\bin\python.exe" free_cad_make_text.py
Fontconfig error: Cannot load default config file
Traceback (most recent call last):
  File "C:\temp\free_cad_make_text.py", line 62, in <module>
    doc.saveAs(file_path)
OSError: {'sclassname': 'class Base::AccessViolation', 'sErrMsg': 'Access violation', 'sfile': '', 'iline': 0, 'sfunction': '', 'swhat': 'Access violation', 'btranslatable': False, 'breported': False}

Roy-043, 2025-01-28:
It is probably better to start forum topic to address this.

Roy-043, 2025-02-23:
I am closing this.

closed 2025-02-23 by Roy-043