FreeCAD/FreeCAD #19274: Memory leak in v1.0

jazzyjoejunior, 2025-01-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I'm experiencing quite a memory leak in FreeCAD v1.0 on a fully updated Win10 22H2 machine with 16GB of RAM, a 12-core processor and 4K monitor. The UI freezes after a few interactions with the menu/functions and a runaway memory leak of 6-15 MB/s and 5% CPU load can be observed in Task Manager next to "FreeCAD main executable". The app throws no error codes and has to be killed. 

The memory leak happens even without opening/working on a model; it is enough to play with the menu or Preferences for a few seconds after launch to trigger it.

I'm observing no memory leaks in v0.21, which runs perfectly smooth on the same machine.

I've tried:
- various Win compatibility modes
- restarting in safe mode 
- newest weekly build
- newest Python

Any help would be appreciated. Thank you.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-27:
Please add steps how to reliably reproduce this.

xtemp09, 2025-01-27:
> 6-15 MB/s

Does it mean that _FreeCAD_ allocates 0.4-1 Gb per minute and does not deallocate, even if you simply go into _Preferences_?

> The app throws no error codes and has to be killed.

When do you decide to terminate the app?

jazzyjoejunior, 2025-01-27:
> Please add steps how to reliably reproduce this.

Dear Max, it is a fresh "out of the box" install with no addons, customizations etc., and the steps to reproduce the problem are pretty simple: I launch the app and the symptoms exhibit every time, usually tens of seconds later. My UI interactions are normal, for example, I click the upper menu bar, followed by Properties, I then click through the properties, and suddenly the UI goes all transparent white and the app freezes. No internal error messages are thrown.

The memory leak observable in Task Manager starts right when the app freezes, sometimes at 6 MB/s, but very often at 15 MB/s. 

Probably a gc.collect() missing somewhere...

Syres916, 2025-01-27:
> I can confirm there is a memory leak going on, in my case I think its more on the order of 3-5MB/s, didn't report it yet since my branch has a bunch of work-in-progerss stuff and wasn't sure if its caused by my changes or not.
> 
> But since someone else is seeing it to, maybe there is something more general going on after all.

Related: https://forum.freecad.org/viewtopic.php?t=94068 ?

jazzyjoejunior, 2025-01-27:
> Does it mean that _FreeCAD_ allocates 0.4-1 Gb per minute and does not deallocate

That seems to be the case. I kill the app a minute or two later because it never comes back to life and the leak keeps going on.

Syres916, 2025-02-01:
Using:

```
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.40041 (Git)
Build type: Debug
Branch: main
Hash: 64a4f6e3669279e5dbe9cb59d8e0123eaeb70833
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
Logical/physical DPI: 96/93.1131
```

I created a leak report from running all the standard Unit Tests as new user with no Addons installed. Hopefully someone with more knowledge than I can determine the set of fixes required and where necessary backport them to 1.0 branch.

[leak-report_FC_1_1_40041.txt.zip](https://github.com/user-attachments/files/18628920/leak-report_FC_1_1_40041.txt.zip)

Syres916, 2025-02-01:
@hyarion are you able to assist, from what I see the biggest leaks are Python specific so I can ignore those, but if you can at least point me through the worst leak that FreeCAD code is responsible for I can probably try to fix some others?

hyarion, 2025-02-03:
I'll see what I can do

hyarion, 2025-02-03:
I've tested a recent main on macos arm64, and I don't see any runaway on my system (rather clean config).
<details><summary>full info</summary>
<p>

OS: macOS Sonoma (14.5)
Architecture: arm64
Version: 1.1.0dev.40041 (Git) Conda
Build type: Release
Branch: main
Hash: 64a4f6e3669279e5dbe9cb59d8e0123eaeb70833
Python 3.13.0, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/Sweden (en_SE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical/physical DPI: 72/127.5
Installed mods: 
  * OpenTheme 2024.9.1

</p>
</details> 

I've checked the log that @Syres916 posted and there's a lot of python and std::map going on, but I believe many of these are false positives, or at least that's what internet tells me when checking some of them out. IIRC it's possible to create a filter to avoid some reoccurring false-positives.

I'm having problems with running leak detection on my own due to programs needing to be signed for inspection `Target binary needs to be debuggable and signed with 'get-task-allow"`, so it's hard for me to inspect this further without falling down the rabbit hole of signing binaries :/

Since we're working with a legacy code base, there's a lot of raw pointers used. At some point we should probably refactor some/most of these to smart pointers, as it's quite easy to miss who's responsible to deallocate raw pointers passed around the system.