FreeCAD/FreeCAD #19279: [Bim/Draft] Expand array is not recalculated correctly without the Always Sync Placement option.

kaiwas, 2025-01-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The multiple array (expand array) is not calculated correctly if the Always Sync Placement option is not activated, even if we update the document. "Mark for recalculation" / F5 and other methods do not help.
Video demonstration.

Video demonstration.

https://github.com/user-attachments/assets/073cae07-5ca0-42f8-9074-9b6d4f193047

A test file is attached (remove the .zip extension).

[Exp_array.FCStd.zip](https://github.com/user-attachments/files/18555956/Exp_array.FCStd.zip)

I understand that expand array is something special. But I assume that it should be recalculated correctly without the forced option.
Expand array is a great way to change some elements of the array. I assume that the recalculation option should somehow take into account manual changes to elements.
In addition, with the Always Sync Placement option, array elements are always marked as required for updating. There should be a way to "recalculate" once.

https://github.com/user-attachments/assets/3c52d405-76f2-4b3f-90b2-d7066db4a609

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39980 (Git) Conda
Build type: Release
Branch: main
Hash: 836ad91f3ddd3ec63b84b8e3f87a767d5465261d
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * FreeCAD_SketchArch
  * lattice2 1.0.0
  * PitchedRoof
  * Road 2025.1.16
  * addFC 1.1.0
  * freecad.gears 1.3.0
  * sheetmetal 0.7.3
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-27:
@Roy-043 FYI

Roy-043, 2025-02-05:
This is probably a Core issue.