FreeCAD/FreeCAD #19284: FreeCAD Sketcher_Constrain Symmetric is not working correctly.

Mito-free, 2025-01-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Sketcher_Constrain Symmetric is not working correctly. In some cases it will not add the constraint to the list at all.

https://github.com/user-attachments/assets/e3a894e3-8556-4e92-92a8-0464f5c71995

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Slovak/Slovakia (sk_SK)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-27:
What is the result when you click first on both endpoints, then the symmetry line (as it is intended)?

Mito-free, 2025-01-27:
The result is still the same regardless of the order.

maxwxyz, 2025-01-27:
Can you try in the latest dev version please? https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Mito-free, 2025-01-27:
Without change. The problem occurs when the center of the arc is X0, Y0, Z0 
as long as the center of the arc is located elsewhere, it's fine.

Syres916, 2025-01-28:
@Mito-free I assume you've cracked the workaround, assume the arc is being created around the X-axis, set the centre point just off the X and Y axes and create the arc as normal, set the symmetrical constraint properly using each end and then the X-axis. Now using the ConstraintPointonObject, select the centre point and the Y-axis to bring the centre point to X0,Y0.

[image]

[image]

[image]

closed 2025-01-29 by maxwxyz